loboris/MicroPython_ESP32_psRAM_LoBo #158: wifi related error message

hutualive, 2018-05-30:
Hi,

after the board connected to wifi AP, occasionally there is error message as below:
--->
E (706988) wifi: esf_buf: t=7 l=222 max:16, alloc:16 no eb
E (706988) wifi: esf_buf: t=7 l=1600 max:16, alloc:16 no eb
E (1007329) wifi: esf_buf: t=7 l=128 max:16, alloc:16 no eb

seems it's wifi event callback related malloc issue.

just bring up the wifi connection and ftp/telnet service. the REPL still running and responding.

it's wrover module flashed with latest firmware.

any idea regard this ?

thanks.

loboris, 2018-05-30:
The message is from the esp-idf wifi stack. It may not be critical, and if everythig runs without other wifi related issues, you can disable wifi error logs with `machine.loglevel('wifi', machine.LOG_NONE)`.

I'll try to investigate it.

closed 2018-05-30 by hutualive

hutualive, 2018-06-01:
@loboris more clues, this time CPU halted. I'm using micropython running on both cores and REPL in separate thread.
--->
 E (46091) wifi: esf_buf: t=7 l=128 max:16, alloc:16 no eb
E (46092) wifi: esf_buf: t=7 l=154 max:16, alloc:16 no eb
E (46092) wifi: esf_buf: t=7 l=154 max:16, alloc:16 no eb
/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/tasks.c:1443 (vTaskDelay)- assert failed!
abort() was called at PC 0x400939dd on core 0

Backtrace: 0x40096dc4:0x3ffe7a90 0x40096f8b:0x3ffe7ab0 0x400939dd:0x3ffe7ad0 0x4            00dbb4d:0x3ffe7af0

CPU halted.