loboris/MicroPython_ESP32_psRAM_LoBo #159: curl module error

hutualive, 2018-05-30:
Hi, 
when using curl module and get the file in example(loboris.eu/ESP32/info.txt), error reported:
E (68365) [Curl]: curl_easy_perform failed: Out of memory

it's latest firmware with wrover module, fresh reboot, import curl and set the curl.options(Verbose: True, Progress: 1, Timeout: 60, Max fsize: 300000, Header len: 1024, Body len: 4096, Decode content: True)

doubled the hdrlen and bodylen, still the same error. 

any idea to tweak ?

thanks.

TribesTom, 2018-05-31:
Hello,

I don't know if it's the same bug but I get a stack overflow with the curl module 

9 times over 10, it crash.
Sometimes I got lucky and it run fine

The error :

`I (1235) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1242) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (1262) [modnetwork]: WiFi Initialized
I (1322) phy: phy_version: 386.0, 67c798f, Mar 14 2018, 16:34:06, 0, 0
D (1324) [modnetwork]: WiFi Started, mode 1
network config: ('192.168.1.150', '255.255.255.0', '192.168.1.1', '192.168.1.1')
Check FIRMWARE
***ERROR*** A stack overflow in task mp_task has been detected.
abort() was called at PC 0x40095b84 on core 1

Backtrace: 0x400959a4:0x3ffbf7e0 0x40095b6b:0x3ffbf800 0x40095b84:0x3ffbf820 0x400920eb:0x3ffbf840 0x400940d8:0x3ffbf860 0x4009408e:0x00000000

CPU halted.`

firmware is compiled with micropython running on the 2 core.

If I disable dual core, It run fine.

Here is the code I run in my main file :

`import program.run,curl,uos

def launch_ota_update():
    import ota
    utime.sleep_ms(20)
    ota.start(server=IPSRV, port=80, file="/fw//MicroPython.bin", restart=True, md5=True)
    utime.sleep_ms(20)

def check_firmware():
    print('Check FIRMWARE')
    versionfile = curl.get(IPSRV+'/fw/version.txt')
    utime.sleep(2)
    print(versionfile)
    if versionfile[0] == 0:
        if "200 OK" in versionfile[1]:
            if not uos.uname().nodename in versionfile[2]:
                print('Firmware need update ... Launching')
                launch_ota_update()
            else:
                print('Firmware already up to date')

check_firmware()

program.run.main()`

hutualive, 2018-06-01:
@TribesTom I'm running micropython on dual core too and REPL in separate thread. I will try with single core to see whether the error can be re-produced or not.

TribesTom, 2018-06-05:
Hello @hutualive 

Did you had time to test ?

Thanks.

hutualive, 2018-06-06:
@TribesTom I have re-built the firmware to run micropython only on App CPU with heap size 2048K, the curl module is not working with default setting. when res = curl.get('loboris.eu/ESP32/info.txt'), the res[0] is not "0", res[1] is valid header and res[2] is error message "Unrecognized or bad HTTP Content or Transfer-Encoding". I tried the ftp protocol(curl.gtp_list()), it's working so I try to change the curl.options as curl.options(verbose=True, hdrlen=2048, bodylen=8192), the same error message but more verbose : * Error while processing content unencoding: Unknown failure within decompression software.
* Closing connection 0
E (283968) [Curl]: curl_easy_perform failed: Unrecognized or bad HTTP Content or Transfer-Encoding

so I change again the curl.options as curl.options(verbose=True, hdrlen=2048, bodylen=8192, nodecode=True), now the curl module working properly. actually you just need set nodecode=True, it will work properly.

@loboris the curl options of nodecode=True is required to get your sample text file, is it possible to auto detect the content encoded or not ?

closed 2018-06-06 by hutualive