loboris/MicroPython_ESP32_psRAM_LoBo #160: wiki doesn't list WiFi events types

carterw, 2018-05-30:
I'm wanting to have a callback invoked when I lose WiFi connection in station mode. The wiki shows how to  register a callback for WiFi events, but I am not seeing a list of the event numbers that might be seen. The example shows a few at the very end, but I am not seeing one that indicates disconnection.

Also, wondering if this feature can be structured such that you don't have to import the entire _network_ module. I've been able to save memory by connecting to WiFi and then deleting the class that imported the _network_ module. With the current implementation you call a method in the _network_ module;

`
network.WLANcallback(wifi_cb)
`

Could it be possible to just import one small class from _network_ that implements the callback, and thereby not have the entire module in memory?

loboris, 2018-06-01:
The callback function receives the tuple argument `(event_ID, event_description, event_info)`
`event_ID` is integer, ESP32 network event number
`event_description` is string describing the event, or None if not available
`event_info` is dictionary providing the events parameters or None if not available

The event numbers are taken from [this structure](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/modnetwork.c#L125), I'll add it to the Wiki.

You must have network module loaded for netwerk to function. It does not take much space (it is a C module and runs directly from Flash).

carterw, 2018-06-01:
Thanks for the info. So "Station disconnected from AP" will be event# 5 and be emitted when we lose WiFi connectivity?

Some of the messages in that list are puzzling. What's the difference between "Ethernet got IP from connected AP" and "Station got IP from connected AP" for example? What is a "Soft-AP"?

loboris, 2018-06-01:
`AP` access point external to ESP32 (when esp32 operates in station mode
`Soft-AP` acccess point (ESP32 itself operating in AP mode)
`Ethernet` ESP32 operates in station mode over Ethernet interface