loboris/MicroPython_ESP32_psRAM_LoBo #164: Using a not connected I2C-object results in RAM leak

kevinkk525, 2018-06-02:
In my program I use I2C to read my HTU which works fine of course.
But if not HTU is connected and I try to read the values the free RAM will decrase. Every 2 reading attempts resulting in "I2C bus error (-1)" cost me 1kB of RAM until the microcontroller crashes with a memory allocation error.
(Garbage collection is executed often between the readings)

carterw, 2018-06-02:
Did you do scan() first to see if it is present?

kevinkk525, 2018-06-02:
No the library just assumes that it is present.
Works on the ESP8266 without the ram leak and just throws an error. I have not seen a htu library yet that scans first.
Also you would have to scan before any read as your connected device could be destroyed and therefore unavailable. That would be a lot of scans. 

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [loboris/MicroPython_ESP32_psRAM_LoBo] Using a not connected I2C-object results in RAM leak (#164)",
"sections": [
{
"text": "",
"activityTitle": "**Bill Carter**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@carterw",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"loboris/MicroPython_ESP32_psRAM_LoBo\",\n\"issueId\": 164,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"loboris/MicroPython_ESP32_psRAM_LoBo\",\n\"issueId\": 164\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/164#issuecomment-394119376"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 341795431\n}"
}
],
"themeColor": "26292E"
}

carterw, 2018-06-02:
Merely suggesting a workaround. I invoke the scan myself first before invoking a driver for a device. An I2C device might be subsequently unplugged, but that's a different problem.

kevinkk525, 2018-06-03:
That's true but I actually just stumbeled onto that bug when I was trying my code on an unconnected esp32 for general workability (import errors, general functionality, ...). 
Never had any problem with it before.

closed 2019-05-01 by kevinkk525