loboris/MicroPython_ESP32_psRAM_LoBo #165: GuruMeditationError (and one other) when trying to start threaded led animations by webServer

crxcv, 2018-06-05:
Hey there 

i have a microWebServer running on esp32 on which u can start some led-animations in a thread. 
but i'm always getting core panics (LoadProhibited) 

if i toggle the animations via touchSensor it works so far! 

u can find the code of the webserver here if needed (not well commented yet - functions used in this issue are handleLightThread, MicroWebSrv.route("/", "POST) and loop at the end of the file): 
https://github.com/crxcv/Leuchteding/blob/master/main.py

led animations here (threads are started at the end of the file - startAnimThread) : 
https://github.com/crxcv/Leuchteding/blob/master/px.py

Error when trying to start rainbowCycle: 
`Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
(PC      : 0x400eb21e  PS      : 0x00060630  A0      : 0x800ea4a7  A1      : 0x3ffe6940  )
PC      : 0x400e1bc3  PS      : 0x00060c30  A0      : 0x800e1e1c  A1      : 0x3ffe6200  

0x400e1bc3: find_qstr at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/qstr.c:136

A2      : 0x00000000  A3      : 0x00000003  A4      : 0x00000000  A5      : 0x3ffcafc4  
A6      : 0x00000001  A7      : 0x00000019  A8      : 0x00000000  A9      : 0x3ffe6930  
A10     : 0x3f411e20  A11     : 0x00000003  A12     : 0x00000000  A13     : 0x3ffcafc4  
A14     : 0xfffffffe  A15     : 0x00000001  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

0x400e1bc3: find_qstr at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/qstr.c:136

0x400e1e19: qstr_data at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/qstr.c:259

0x400f0e2a: mp_obj_str_binary_op at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objstr.c:318 (discriminator 1)

0x400ea4b1: mp_binary_op at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:540

0x400f6ce7: mp_execute_bytecode at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1262

0x400ede5e: fun_bc_call at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:283

0x400e9589: mp_call_function_n_kw at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:601

0x400f53e5: thread_entry at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/modthread.c:121

0x400d8c4d: freertos_entry at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/mpthreadport.c:612

Backtrace: 0x400eb21e:0x3ffe6940 0x400ea4a4:0x3ffe6960 0x400f6ce7:0x3ffe69a0 0x400ede5e:0x3ffe6a40 0x400e9589:0x3ffe6a70 0x400f53e5:0x3ffe6a90 0x400d8c4d:0x3ffe6b30`

error when trying to start fire animation: 
`assertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "/home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/gc.c", line 596, function: gc_free
abort() was called at PC 0x40106a67 on core 1

0x40106a67: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40090be8:0x3ffe67d0 0x40090d8b:0x3ffe67f0 0x40106a67:0x3ffe6810 0x400e163a:0x3ffe6840 0x400e0e7d:0x3ffe6860 0x400ede95:0x3ffe6880 0x400e9589:0x3ffe68b0 0x400f68f1:0x3ffe68d0 0x400ede5e:0x3ffe6970 0x400e9589:0x3ffe69a0 0x400f53e5:0x3ffe69c0 0x400d8c4d:0x3ffe6a60

0x40090be8: invoke_abort at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/panic.c:648

0x40090d8b: abort at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/panic.c:648

0x40106a67: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400e163a: gc_free at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/gc.c:596 (discriminator 1)

0x400e0e7d: m_free at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/malloc.c:188

0x400ede95: fun_bc_call at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:328

0x400e9589: mp_call_function_n_kw at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:601

0x400f68f1: mp_execute_bytecode at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:938

0x400ede5e: fun_bc_call at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:283

0x400e9589: mp_call_function_n_kw at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:601

0x400f53e5: thread_entry at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/modthread.c:121

0x400d8c4d: freertos_entry at /home/chrissi/coding/python/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/mpthreadport.c:612
`

would be gread if someone could help - is the first time i'm really working with threads!

loboris, 2018-06-05:
I'll test your code to find the possible problem,
On a first look, I've found one potentional isue [here](https://github.com/crxcv/Leuchteding/blob/master/main.py#L33)

`_thread.notify(lightAnim_thread, _thread.EXIT) `and `_thread.stop(lightAnim_thread)` are doing the same thing and should not both be used.

crxcv, 2018-08-30:
seems to be caused by memory fragmentation 

memory after crash: 

`>>> micropython.mem_info(1)
stack: 768 out of 15360
GC: total: 80000, used: 42992, free: 37008
 No. of 1-blocks: 356, 2-blocks: 59, max blk sz: 383, max free sz: 381
GC memory layout; from 3ffc4b30:
00000: MDMhhhDhhShhShShq=======Q=======================================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: ================================================================
01400: ================================================================
01800: =======================MhhBTSSBDhhSShMDSShSMDSSMDhBSSh=Shhh==hhh
01c00: Fh=FSh=Sh=SSh=Sh=SSSSShh===h=TBThhhLBBhhSq=======SSSSSSSSSq=====
02000: ==SSSSSShSSq=======hhhMDh=hhhTFSShh=TMDBSh==LSFFShhhSSSFFFFFFFFS
02400: h===h===FFFFFFFFh=SFFFh==SSh======Shh=Sh=h=============Sh=h=====
02800: =SSShSSq=======h==============h=========Shh==h=SSh====h=======Sh
02c00: =h======================h==h===============SSSSSh=BTTBTDh=DBBBhB
03000: hSh=h==================h==Shh============hh=h==h=h===Bh==Dhh====
03400: ==h========hh=Bhq=======h======hSh=====Sh=h===========hhh=h=====
03800: ==hq=======h=========ShhhhSh=hh=h===hh==h=============Sh=h======
03c00: ======================h================FFSh==============h======
04000: =======h==============h============h============h===============
04400: ===========h=============TB=h===hh===BBBhB=BhBh===========h=====
04800: Sh=h===========h===========h=========hhSh=h================h=q==
04c00: ================hq======SSShq===========BBBBTBhBhB=hB=hB=BBhB=h=
05000: hB=B=BBB=BB==BBBBBh====h===hh=BBhhhhhhhhhDTBhhh=h===B=BBBh======
05400: =====h=====Bh===h==q=======BBBBBh==============hSh=BBh===h=h==h=
05800: Bhh========h============h=h=hh=Sh=B.....h=....h=.......h........
05c00: .......Sh=h=====...hhh====h==========h====h=============h=======
06000: h=========.........................................h============
06400: =====================h==========h======h=====h================h=
06800: ================================================================
06c00: ===......hhhhh==hh.....hq=========...............h==============
07000: =========..........................hh=====================......
       (2 lines all free)
07c00: ..................hq=======.....................................
08000: ..................h===========================..................
       (2 lines all free)
08c00: ...........................................................hq===
09000: ====...............hhh==h.....hhhh=.hhhhh.....hh...hhh...h.....h
09400: ...................................................h======h=====
09800: =====..hq=======................................................
       (2 lines all free)
0a400: h====h========h================h=======h===========h============
0a800: =h=====..........................q=======h======================
0ac00: =========================.......................................
       (3 lines all free)
0bc00: .............q=================================.................
0c000: ................................................................
0c400: .....h.....hhh..................................................
       (2 lines all free)
0d000: ..............hh...h............................................
0d400: .......................................................hh......h
0d800: hh........................h=....................................
0dc00: ................................................................
0e000: .............h==========================h=====h=====h===h====h==
0e400: h=====h=========h======h======hh=========hh=============hh=====h
0e800: h====hh==========h=h===================hh=========h=h===========
0ec00: =hh==============h=h=============hh=================h===========
0f000: ===============hh==h================hh==hh===hh==hh===hh==hh===h
0f400: h======h=h==hh==hh=hh==hh===hh==hh===hh===hh===hh===hh==h.......
       (5 lines all free)
10c00: ..h=============================================================
11000: ================================================================
11400: ================================================================
11800: ================================================================
11c00: ================================================================
12000: ===========.....................................................
       (5 lines all free)
13800: ........
`

i'll close this!

closed 2018-08-30 by crxcv