loboris/MicroPython_ESP32_psRAM_LoBo #167: Buld firmware under windows (msys2) failed

df6dbf, 2018-06-07:
Build Firmware fails since last update. Started mingw32.exe changed the Directorys as described. git pull to get the latest repository. 

./BUILD.sh clean is ok
./BUILD.sh menuconfig also runs without any error.
./BUILD.sh all fails - see attached screenshot from my win7pro-64bit machine.

Kind regards, 
Holger
[image]

EliSu, 2018-06-07:
How about move the msys32 folder to D:\， then retry?

df6dbf, 2018-06-08:
I dont have a Drive D on this machine. :)  The odd thing is that die building process worked before the last update. seems ths file stdint.h is missing.

df6dbf, 2018-06-09:
Finally installed the toolchain and the respository completly new. Now it works.

closed 2018-06-09 by df6dbf