loboris/MicroPython_ESP32_psRAM_LoBo #169: mpy-cross compiled module fail to load on ESP32

faisyl, 2018-06-12:
Are there any specific options needed to get the mpy-cross to create loadable modules for your version of MicroPython? I tried with the tool created under components/mpy_cross_build but it fails to import on the ESP32. Here's the traceback:

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: incompatible .mpy file
```

faisyl, 2018-06-14:
Found the problem. I had built micropython with UTF-8 support switched off, but the tool was generating code with UTF-8 enabled in its header, so there was a mismatch between the runtime and the compiled code code magic strings. I wish there was an easier way to sync this with the actual configuration.

closed 2018-06-14 by faisyl

saramonteiro, 2019-10-23:
So, how did you enable it? I am having the same error. I used mpy-cross with the following command:
`python -m mpy_cross -s -march=xtensa mycode.py`

curlyz, 2020-06-05:
Really it's about different implementation may have different flag. Once compiled with mpy-cross, just read out its first 3 bytes . [0] = 'M'. , [1] = 0x03 and [2] is from 0x00 to 0x03 ( you can get which one in sys.implementation, or just brute force it, there is only 4 possible value of it). Replace those bytes and you got yourself a functional mpy. Note that this is not always work well.