loboris/MicroPython_ESP32_psRAM_LoBo #172: MQTT client doesn't work with rabbitmq

thijstriemstra, 2018-06-14:
When I use the `network.mqtt` client it fails to publish data to a rabbitmq broker with the mqtt plugin enabled.

Error on esp32:

```
----------------------------------------
Connecting to broker: 192.168.42.10
Connection: OK
----------------------------------------

* Light: 63.3333333333333 on topic 'devices/foo/light/lux'
E (22147) MQTT_CLIENT: Error write data or timeout, written len = -1
E (22149) MQTT_CLIENT: Error publishing data to topic=devices/foo/light/lux, qos=0
```

rabbitmq log:

```
=INFO REPORT==== 13-Jun-2018::23:36:48 ===
accepting MQTT connection <0.667.0> (192.168.42.15:53878 -> 192.168.42.10:1883)

=INFO REPORT==== 13-Jun-2018::23:36:50 ===
closing MQTT connection <0.667.0> (192.168.42.15:53878 -> 192.168.42.10:1883)
```

https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py works fine though.

thijstriemstra, 2018-06-14:
Oh, this is probably a duplicate of #171 opened 4 hrs earlier ;)

loboris, 2018-06-26:
Please see this [comment](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/171#issuecomment-400316248)

thijstriemstra, 2018-06-26:
I'm running a local rabbitmq server (3.6.x) and not able to provide a remote test server. Since others also report the issue with mosquitto I'll wait till that issue will get resolved somehow. Thanks for the feedback though.

loboris, 2018-06-26:
I cannot reproduce the issue comnnecting to my local mosquito server (on Ubuntu 17.10) to.

I don't know how to resolve the issue if I can't reproduce it. Could you, at least, give some more information about your mqtt server instalation (platform, OS, configuration, eetc).

thijstriemstra, 2018-06-26:
Operating system is Ubuntu 16.04. Rabbitmq version 3.6. Install rabbitmq using the APT repository or .deb file on https://www.rabbitmq.com/install-debian.html

Enable MQTT: https://www.rabbitmq.com/mqtt.html (basically as simple as:

```
sudo rabbitmq-plugins enable rabbitmq_mqtt
```

And you're good to go! If you need any more info, please ask, thanks.

loboris, 2018-06-26:
I have installed **rabbitmq** on Ubuntu 16.04 (running in virtual machine).
Enabled mqtt, added user `boris`.
No issues when running ESP32 mqtt client:
```python
import network

def conncb(task):
    print("[{}] Connected".format(task))

def disconncb(task):
    print("[{}] Disconnected".format(task))

def subscb(task):
    print("[{}] Subscribed".format(task))

def pubcb(pub):
    print("[{}] Published: {}".format(pub[0], pub[1]))

def datacb(msg):
    print("[{}] Data arrived from topic: {}, Message:\n".format(msg[0], msg[1]), msg[2])

>>> rmqtt = network.mqtt("local", "mqtt://192.168.0.33", user="boris", password="boris", cleansession=True, connected_cb=conncb, disconnectermqtt.start()
>>> D (4743561) TRANS_TCP: [sock=62],connecting to server IP:192.168.0.33,Port:1883...
[local] Connected

>>> rmqtt.subscribe('test')
True
>>> [('local', 'test')] Subscribed

>>> rmqtt.publish('test', 'Hi from Micropython')
True
>>> [local] Data arrived from topic: test, Message:
 Hi from Micropython

>>> 
```

rabbitmq log:
```
=INFO REPORT==== 26-Jun-2018::16:36:43 ===
Starting RabbitMQ 3.5.7 on Erlang 18.3
Copyright (C) 2007-2015 Pivotal Software, Inc.
Licensed under the MPL.  See http://www.rabbitmq.com/

=INFO REPORT==== 26-Jun-2018::16:36:43 ===
node           : rabbit@ubuntu-server
home dir       : /var/lib/rabbitmq
config file(s) : /etc/rabbitmq/rabbitmq.config (not found)
cookie hash    : hCDgsHsJpTl1J+ioXLI/hQ==
log            : /var/log/rabbitmq/rabbit@ubuntu-server.log
sasl log       : /var/log/rabbitmq/rabbit@ubuntu-server-sasl.log
database dir   : /var/lib/rabbitmq/mnesia/rabbit@ubuntu-server
...
...
=INFO REPORT==== 26-Jun-2018::16:52:16 ===
accepting MQTT connection <0.567.0> (192.168.0.26:55226 -> 192.168.0.33:1883)

```

thijstriemstra, 2018-06-26:
That's very weird.. I really want to take advantage of your C-implementation but I'm not sure what's wrong with my setup then.. I will try again on the weekend and let you know if you can create a better reproducable example.

slzatz, 2018-06-26:
Could there be an issue when no password is required or have you also tested that.  It's a no password broker where it stopped working for me.

thijstriemstra, 2018-06-27:
@slzatz might be onto something here @loboris, I also didn't use a username/password in my setup. Does it work for you with an anonymous connection?

loboris, 2018-06-27:
Yes it works without issues without username/password (you can try with `iot.eclipse.org`).

This issue may be related to [Error write data or timeout #48](https://github.com/tuanpmt/espmqtt/issues/48)

The repository will be updated today or tomorrow to include the updated **esp-idf** needed for it to work.

When updated, try to build with
`→ Component config → LWIP → Support LWIP socket select() only` (`CONFIG_USE_ONLY_LWIP_SELECT`)
**set** in **menuconfig** to see if there is any difference.

It is still strange that I never could reproduce this issue.

slzatz, 2018-06-28:
While this is less than satisfying as to an explanation of what was going on -- I upgraded to today's commit and **without changing anything in menuconfig**, mqtt is now working for me.

thijstriemstra, 2018-07-04:
Tested again with master and it works now:

```python
>>> def conncb(task):
...     print("[{}] Connected".format(task))
>>> mqtt = network.mqtt('local', 'mqtt://192.168.2.8', user='foo', password='bar', cleansession=True, connected_cb=conncb)
>>> mqtt.start()
>>> [local] Connected
mqtt.publish('test', 'Hi from Micropython')
True
```

closed 2018-07-04 by thijstriemstra