loboris/MicroPython_ESP32_psRAM_LoBo #174: rtc.wake_on_ext1() won't accept tuple

carterw, 2018-06-18:
on the wiki, in the RTC class section the documentation shows;

**rtc.wake_on_ext1(pins [,level])**
**Enable external interrupt #1 on multiple pins.**

**pins tuple of Pin objects to be used as wakeup source: (Pin(x), Pin(y), ..., Pin(z))**

On the main README page it shows;
**rtc.wake_on_ext0(Pin, level)**
**rtc.wake_on_ext1(Pin, level)**
  **# wake up from deepsleep on pin level**

Presumably the README is wrong, wake_on_ext1 expects a tuple of Pin objects? However I am trying to set a wakeup on just one pin and getting obscure error messages regardless of the arguments i use. For example;

`pObj = machine.Pin(0)`
`machine.RTC.wake_on_ext1((pObj,), 0)`
TypeError: argument has wrong type

carterw, 2018-06-25:
Can someone please verify this? I would like to use ext1, maybe I am just not coding it right.

loboris, 2018-06-26:
Use it this way (create machine.RTC instance object first):
```
>>> import machine
>>> rtc = machine.RTC()
>>> pObj = machine.Pin(0)
>>> rtc.wake_on_ext1((pObj,), 0)
>>> 
```
You can also use the pin numbers instead of Pin objects
```
>>> rtc.wake_on_ext1((0,), 0)
>>> rtc.wake_on_ext1((0,12,13), 0)
>>> 
```

closed 2018-06-26 by carterw