loboris/MicroPython_ESP32_psRAM_LoBo #178: Display module: Touch is not reliable

amirgon, 2018-06-25:
I'm using a TFT ILI9341 module with XPT2066 touch controller, like this one:
[image]

While the display works fine and a touch can be detected, the touch coordiates do not make sense and are not reliable.  
This is not a calibration issue - a press on the same spot returns totally different results, sometimes with X and Y coordinates swapped.  
In the example below I touched the **same** spot after each call to `tft.gettouch`. Every touch caused `tft.gettouch` to return a result, but the returned coordinates do not make sense:

```
>>> import display
>>> tft = display.TFT()
>>> tft.init(tft.ILI9341, width=240, height=320, miso=19, mosi=18, clk=5, cs=15, dc=33, bgr=True, hastouch=tft.TOUCH_XPT, tcs=32)
>>> tft.gettouch(raw=1,wait=50000)
(True, 0, 529)
>>> tft.gettouch(raw=1,wait=50000)
(True, 518, 8)
>>> tft.gettouch(raw=1,wait=50000)
(True, 0, 288)
>>> tft.gettouch(raw=1,wait=50000)
(True, 7, 264)
>>> tft.gettouch(raw=1,wait=50000)
(True, 0, 264)
>>> tft.gettouch(raw=1,wait=50000)
(True, 270, 32)
>>> tft.gettouch(raw=1,wait=50000)
(True, 7, 0)
>>> tft.gettouch(raw=1,wait=50000)
(True, 15, 0)
>>> tft.gettouch(raw=1,wait=50000)
(True, 264, 0)
>>> tft.gettouch(raw=1,wait=50000)
(True, 0, 17)
>>> tft.gettouch(raw=1,wait=50000)
(True, 520, 0)
```

loboris, 2018-06-26:
I have tested with the same display couple of months ago, and the touch was quite reliable.
After calibrating with `tpcalib.py` the demo application `components/micropython/esp32/modules_examples/tft/paint.py` was running without issues.
I'll check if I've introduced some bug in the latest versions.

taitix, 2018-10-24:
I have similar issue with a red ILI9341 module with XPT module.
It seems this is a software issue since examples from the following arduino library works fine: https://github.com/Bodmer/TFT_eSPI

I managed to get meaningful raw (z, x, y) values by following change (orientation is messed up though...).
https://github.com/taitix/MicroPython_ESP32_psRAM_LoBo/commit/d93202b1795f84f6ff5a4f681154f0e73c908419
Because this is the only touch device I own,  I'm not sure if this is device specific behavior or some bug which affect all the device.

masterofhw, 2018-11-22:
I think, there is a problem in tftpspi.c. The bits returning from xpr2046 are in big endian order, so you have to swap the bytes. I have made folling changes and it work for me:

@@ -604,8 +607,9 @@ int touch_get_data(uint8_t type)
        esp_err_t ret = spi_transfer_data_nodma(ts_spi, &t);
        if (ret != ESP_OK) return 0;
 
-    uint16_t res = (uint16_t)(buf >> 8);
-
+    uint16_t res = ((buf>>16)&0xff)+(buf&0xff00);
+    res=(res>>3) & 0xfff;
+    
     return res;
 }

derkomai, 2019-03-26:
> I think, there is a problem in tftpspi.c. The bits returning from xpr2046 are in big endian order, so you have to swap the bytes. I have made folling changes and it work for me:
>
> @@ -604,8 +607,9 @@ [...]

Do you mean a XPT2046? How are you initializating the display and callibrating it? I'm trying your changes, but I can't seem to get any touch response.