loboris/MicroPython_ESP32_psRAM_LoBo #179: Hi boris

elektrobil, 2018-06-27:
I need some help about WIFI Stations. When connecting a client on ESP32 over DHCP it works good but when I pugged off power of client ESP32 wifi doesnt sense disconnected client. I try with my smart phone connect and disconnect ESP32 senses disconnecting My phone. How could I refresh the client list or inactive timer for wifi stations ?
Im using this command for listing clients 
`wifi.status('stations')`

loboris, 2018-06-27:
I'm not shure if understand the problem.
Are you starting AP mode on ESP32, like this:
```python
>>> import network
>>> ap_if = network.WLAN(network.AP_IF)
>>> ap_if.active(True)
True
```
Using `wifi.status('stations')` you can get the list of connected stations:
```python
>>> ap_if.status('stations')
[]
```
no station connected, then when I connect with my phone:
```python
>>> ap_if.status('stations')
[(b"\x00'\x158i\xcc", '192.168.4.2')]
```
After I disconnect the phone:
```python
>>> ap_if.status('stations')
[]
```

elektrobil, 2018-06-27:
I added a client board on ESP32 when I plugged off power of client still appear in list

elektrobil, 2018-06-27:
Sorry Boris
Im closing issu for going on forum

closed 2018-06-27 by elektrobil