loboris/MicroPython_ESP32_psRAM_LoBo #182: Pin.irq is now Pin.irqvalue ?

carterw, 2018-06-28:
Did the Pin class change? Breaks existing code.

In firmware prior to today's commit;
>>> from machine import Pin
>>> help(Pin)
object \<class 'Pin'\> is of type type
  init -- \<function\>
  value -- \<function\>
  irq -- \<function\>

In firmware built with today's commit;
>>> from machine import Pin
>>> help(Pin)
object <class 'Pin'> is of type type
  __del__ -- \<function\>
  init -- \<function\>
  value -- \<function\>
  irqvalue -- \<function\>           <<<<

carterw, 2018-06-28:
Actually it looks like Pin.irq is simply missing. Changing it to Pin.irqvalue results in;
TypeError: function does not take keyword arguments

I guess Pin.irqvalue() is just a new way to see how the irq is currently set up.

loboris, 2018-06-28:
Yes, the machine.Pin was refactored, see the [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/pin).
I think it makes much more sense that way, and many more features are anailable.
It was changed in commit from June 4, not in todays commit.

carterw, 2018-06-28:
Wow, okay. Will investigate that.

closed 2018-06-28 by carterw