loboris/MicroPython_ESP32_psRAM_LoBo #183: Kernel panic with MicroWebSrv

TribesTom, 2018-06-29:
Hello Loboris,

I have some strange problem with MicroWebSrv.

When I launch the MicroWebSrv, I have either a kernel panic or another thread crashing 

1 case : 
`>>>
>>> from   microWebSrv import MicroWebSrv
>>>
>>> import _thread
>>> _thread.list()
ID=1073565532, Name: BME280, State: running, Stack=6144, MaxUsed=1900, Type: PYTHON
ID=1073565148, Name: MQTT, State: waiting, Stack=6144, MaxUsed=2320, Type: PYTHON
ID=1073428980, Name: MainThread, State: running, Stack=20480, MaxUsed=8480, Type: MAIN
>>> Send to thread 1073565148 data : {"humidity": "39.62", "temperature": "25.27", "pressure": "981.25"}
publish to test/bureau/microsensor/test1 msg : {"humidity": "39.62", "temperature": "25.27", "pressure": "981.25"}

>>>
>>> srv = MicroWebSrv()
>>> srv.Start(threaded=True, stackSize=16384)
>>> Send to thread 1073565148 data : {"humidity": "40.14", "temperature": "24.96", "pressure": "981.23"}
publish to test/bureau/microsensor/test1 msg : {"humidity": "40.14", "temperature": "24.96", "pressure": "981.23"}
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ef90e  PS      : 0x00060730  A0      : 0x800ef968  A1      : 0x3ffd6840
A2      : 0x3f818880  A3      : 0x0000036e  A4      : 0x3ffd6934  A5      : 0x00000000
A6      : 0xfffffffe  A7      : 0x00000001  A8      : 0x0000002e  A9      : 0x3ffd6820
A10     : 0x00000000  A11     : 0x3ffd6938  A12     : 0x00000001  A13     : 0x3f40ed3f
A14     : 0x3f40ed47  A15     : 0x00000009  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000001c  LBEG    : 0x400e7288  LEND    : 0x400e72b4  LCOUNT  : 0x0002e96e

Backtrace: 0x400ef90e:0x3ffd6840 0x400ef965:0x3ffd6860 0x400fcdfd:0x3ffd6880 0x400f4482:0x3ffd6920 0x400ef7b9:0x3ffd69a0 0x400fc1a1:0x3ffd69c0 0x400dab35:0x3ffd6a60

CPU halted.` 
2nd case :
`>>>
>>>
>>> from   microWebSrv import MicroWebSrv
>>> srv = MicroWebSrv()
>>> srv.Start(threaded=True, stackSize=16384)
>>>
>>>
>>>
>>> import _thread
>>> _thread.list()
ID=1073586912, Name: MicroWebServer, State: running, Stack=16384, MaxUsed=1408, Type: PYTHON
ID=1073565532, Name: BME280, State: running, Stack=6144, MaxUsed=1788, Type: PYTHON
ID=1073565148, Name: MQTT, State: waiting, Stack=6144, MaxUsed=2320, Type: PYTHON
ID=1073428980, Name: MainThread, State: running, Stack=20480, MaxUsed=8384, Type: MAIN
>>> Send to thread 1073565148 data : {"humidity": "40.10", "temperature": "25.17", "pressure": "981.23"}
publish to test/bureau/microsensor/test1 msg : {"humidity": "40.07", "temperature": "25.16", "pressure": "981.21"}
Unhandled exception in thread started by <function run at 0x3f815160>
Traceback (most recent call last):
  File "program/mqttha.py", line 26, in run
AttributeError: 'int' object has no attribute 'publish'

>>>
>>> _thread.list()
ID=1073586912, Name: MicroWebServer, State: running, Stack=16384, MaxUsed=1408, Type: PYTHON
ID=1073565532, Name: BME280, State: running, Stack=6144, MaxUsed=1964, Type: PYTHON
ID=1073428980, Name: MainThread, State: running, Stack=20480, MaxUsed=8384, Type: MAIN
>>>`

I have only the example index.html in the www directory.

It seems the microserver thread is accessing some memory it shouldn't.

If you need other example or the code to test I can try to plublish it.

Thanks

TribesTom, 2018-06-30:
Another test this morning.

I launched the webserver before the others threads this time

`>>> Send to thread 1073607652 data : {"humidity": "52.06", "temperature": "22.51", "pressure": "980.91"}
publish to test/bureau/microsensor/test1 msg : {"humidity": "52.08", "temperature": "22.52", "pressure": "980.91"}
***ERROR*** A stack overflow in task MicroWebServer has been detected.
abort() was called at PC 0x40095d88 on core 1
Backtrace: 0x40095ba8:0x3ffda170 0x40095d6f:0x3ffda190 0x40095d88:0x3ffda1b0 0x4009172a:0x3ffda1d0 0x40093944:0x3ffda1f0 0x400938fa:0x3ffda1f8
CPU halted.
`
I have tested with 4k and 8k stack in thread launch.

It crash when i try to load the index.html example in the browser.

TribesTom, 2018-06-30:
Hello,

Here is a link with the code : 

[https://git.open-nt.be/TribesTom/MicroSensor-pub](https://git.open-nt.be/TribesTom/MicroSensor-pub)

The part about the ota is disabled as I m not using an OTA firmware for now.
The progupdate is the same as the ota but for the py code.

After this is the real program : 

I have 1 thread for a BME280 sensor.
1 thread for the MQTT which receive notification from BME thread
And one thread for the webserver which will be used later for updating the configuration.

Thanks

loboris, 2018-06-30:
I'll try to test the code next week.

crxcv, 2018-08-28:
also still struggeling with similar issue with LoadProhibited
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/165

i have neopixel animations running in a thread. as soon as i try to change sth with the webserver i got that error.

after alot of reading i checked the sizes of the threads with _thread.list() and found out that the pixel animatios are very memory consuming. means that max_used was always just a bit smaller than max_size of the thread. 
so i increased the stack size of the pixel animations..
now guruMeditationError is almost gone but ended up with another error: 
`assertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "/home/chrissi/coding/microPy/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/gc.c", line 699, function: gc_realloc`

crxcv, 2018-08-30:
hey guys,
i don't think this is an issue.
Did u ever think of fragmentation? 

i implemented the whole stuff with the original microPy framework.
When trying to create a Thread i got an error - memory allocation failed. So i took a look on the memory.

Thats what my memory looks like after crash: 

`>>> micropython.mem_info(1)
stack: 768 out of 15360
GC: total: 80000, used: 42992, free: 37008
 No. of 1-blocks: 356, 2-blocks: 59, max blk sz: 383, max free sz: 381
GC memory layout; from 3ffc4b30:
00000: MDMhhhDhhShhShShq=======Q=======================================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: ================================================================
01400: ================================================================
01800: =======================MhhBTSSBDhhSShMDSShSMDSSMDhBSSh=Shhh==hhh
01c00: Fh=FSh=Sh=SSh=Sh=SSSSShh===h=TBThhhLBBhhSq=======SSSSSSSSSq=====
02000: ==SSSSSShSSq=======hhhMDh=hhhTFSShh=TMDBSh==LSFFShhhSSSFFFFFFFFS
02400: h===h===FFFFFFFFh=SFFFh==SSh======Shh=Sh=h=============Sh=h=====
02800: =SSShSSq=======h==============h=========Shh==h=SSh====h=======Sh
02c00: =h======================h==h===============SSSSSh=BTTBTDh=DBBBhB
03000: hSh=h==================h==Shh============hh=h==h=h===Bh==Dhh====
03400: ==h========hh=Bhq=======h======hSh=====Sh=h===========hhh=h=====
03800: ==hq=======h=========ShhhhSh=hh=h===hh==h=============Sh=h======
03c00: ======================h================FFSh==============h======
04000: =======h==============h============h============h===============
04400: ===========h=============TB=h===hh===BBBhB=BhBh===========h=====
04800: Sh=h===========h===========h=========hhSh=h================h=q==
04c00: ================hq======SSShq===========BBBBTBhBhB=hB=hB=BBhB=h=
05000: hB=B=BBB=BB==BBBBBh====h===hh=BBhhhhhhhhhDTBhhh=h===B=BBBh======
05400: =====h=====Bh===h==q=======BBBBBh==============hSh=BBh===h=h==h=
05800: Bhh========h============h=h=hh=Sh=B.....h=....h=.......h........
05c00: .......Sh=h=====...hhh====h==========h====h=============h=======
06000: h=========.........................................h============
06400: =====================h==========h======h=====h================h=
06800: ================================================================
06c00: ===......hhhhh==hh.....hq=========...............h==============
07000: =========..........................hh=====================......
       (2 lines all free)
07c00: ..................hq=======.....................................
08000: ..................h===========================..................
       (2 lines all free)
08c00: ...........................................................hq===
09000: ====...............hhh==h.....hhhh=.hhhhh.....hh...hhh...h.....h
09400: ...................................................h======h=====
09800: =====..hq=======................................................
       (2 lines all free)
0a400: h====h========h================h=======h===========h============
0a800: =h=====..........................q=======h======================
0ac00: =========================.......................................
       (3 lines all free)
0bc00: .............q=================================.................
0c000: ................................................................
0c400: .....h.....hhh..................................................
       (2 lines all free)
0d000: ..............hh...h............................................
0d400: .......................................................hh......h
0d800: hh........................h=....................................
0dc00: ................................................................
0e000: .............h==========================h=====h=====h===h====h==
0e400: h=====h=========h======h======hh=========hh=============hh=====h
0e800: h====hh==========h=h===================hh=========h=h===========
0ec00: =hh==============h=h=============hh=================h===========
0f000: ===============hh==h================hh==hh===hh==hh===hh==hh===h
0f400: h======h=h==hh==hh=hh==hh===hh==hh===hh===hh===hh===hh==h.......
       (5 lines all free)
10c00: ..h=============================================================
11000: ================================================================
11400: ================================================================
11800: ================================================================
11c00: ================================================================
12000: ===========.....................................................
       (5 lines all free)
13800: ........
`