loboris/MicroPython_ESP32_psRAM_LoBo #187: Pin interrupt causes stack overflow

untitledlt, 2018-07-13:
Not sure how to trace what exactly causes this, but after firing few hundreds interrupts i'm getting this:

```
***ERROR*** A stack overflow in task pin_task has been detected.
abort() was called at PC 0x400923c4 on core 1

Backtrace: 0x40092208:0x3ffd89f0 0x400923ab:0x3ffd8a10 0x400923c4:0x3ffd8a30 0x4008e3d8:0x3ffd8a50 0x400902c4:0x3ffd8a70 0x4009027a:0xa5a5a5a5

CPU halted.
```

Using latest ESP32_LoBo_v3.2.20.
Did not had this issue with previews versions.

loboris, 2018-07-15:
Thanks for reporting the issue. I'll check the debounce task to see what could cause the issue.

untitledlt, 2018-07-16:
I have managed to fix debounce value check and i have build new firmware but still have problems with interrupts. Sometimes CPU halts, sometimes callback just stops firing without any errors.
Here is a minimal code to reproduce:

```
from machine import Pin

led = Pin(2, Pin.OUT, Pin.PULL_DOWN)
ledValue = False

def cb(p):
    global ledValue
    ledValue = not ledValue
    led.value(ledValue)

pin = Pin(13, Pin.IN, Pin.PULL_DOWN, trigger=Pin.IRQ_ANYEDGE, handler=cb)
```

vrubel, 2018-07-16:
You did not try to use the approach described here? 
http://docs.micropython.org/en/latest/pyboard/reference/isr_rules.html#isr-rules

loboris, 2018-07-16:
@untitledlt
Are you using a button to cause the pin interrupt or something else?
Pin interrupts are not intended to track the fast pin changes, anything more frequent than 100 Hz can cause issues.
You may try to use the **level** interrupts. The interrupts will be disable before the callback function is executed, and must be enabled in the callback function (possible changing the interrupt level).

I'll try to do some more testing on Pin interrupts later this week.

untitledlt, 2018-07-16:
@loboris are you talking about `IRQ_LOLEVEL` and `IRQ_HILEVEL`? 
Not sure how to use them as direct replacement does not work.
I'm using interrupts for buttons but with very bad mechanical contacts so possibly there are a lot of interrupts happening.

Anyway, i tried the following code on Micropython build from micropython.org and no matter how hard i tried, i wasn't able to crash CPU or some other way stop it responding.

```
from machine import Pin

ledValue = False

def cb(p):
    global ledValue
    ledValue = not ledValue
    led.value(ledValue)

led = Pin(2, Pin.OUT, Pin.PULL_DOWN)
pin = Pin(13, Pin.IN, Pin.PULL_DOWN)
pin.irq(trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING, handler=cb)
```

untitledlt, 2018-07-16:
Can't find any examples how to use `IRQ_HIGH_LEVEL`/`IRQ_LOW_LEVEL`.
Could please someone provide minimal working example how to use it?

nitruxa, 2018-08-02:
@loboris when do u think it can be addressed?

untitledlt, 2018-08-28:
any progress on this issue?

loboris, 2018-08-28:
I'm, checking the Pin module right now, the update can be expected in a couple of days.

nitruxa, 2018-09-06:
is ir ready already?

loboris, 2018-09-06:
The update was commited couple of days ago.
I'm still working on testing all possible configuration combinations...

nitruxa, 2018-10-10:
@loboris is this one fixed already?

capedra, 2018-10-11:
@nitruxa @loboris 
@untitledlt 

```
Pressed pushbutton at 80.342391
***ERROR*** A stack overflow in task pin_task has been detected.
abort() was called at PC 0x400983ac on core 1

Backtrace: 0x4009823b:0x3ffedca0 0x40098393:0x3ffedcc0 0x400983ac:0x3ffedce0 0x400949b5:0x3ffedd00 0x40096724:0x3ffedd30 0x400966da:0x00000000

CPU halted.
```

capedra, 2018-10-11:
@loboris 
I dealt with the de-bounce issue using the following link. The issue is that the following script only works with the original firmware from micropython.org because irq function doesn't exist anymore (irqvalue doesn't work with it):
https://gist.github.com/SpotlightKid/0a0aac56606286a80f860b116423c94f

How may we deal with it? Thanks!