loboris/MicroPython_ESP32_psRAM_LoBo #192: SD Mode = SPI Issue

bdespatis, 2018-07-20:
Hi,

I found a problem with my electric engineer Gregory. We can't use our SD card in SPI Mode with the new firmware. Last April, I had no issues and it was working great (same line of code)
```` 
>>> import os, uos
>>> uos.sdconfig(uos.SDMODE_SPI,clk=27,mosi=26,miso=5,cs=18)
>>> os.mountsd()
---------------------
 Mode: SPI
 Name: SA04G
 Type: SDHC/SDXC
Speed: default speed (25 MHz)
 Size: 3796 MB
  CSD: ver=1, sector_size=512, capacity=7774208 read_bl_len=9
  SCR: sd_spec=2, bus_width=5
````
Now it looks like this : 
````
>>> import os
>>> import uos
>>> uos.sdconfig(uos.SDMODE_SPI,clk=27, mosi=26,miso=5,cs=18)
>>> os.mountsd()
E (65592) sdmmc_cmd: sdmmc_card_init: send_scr (2) returned 0x107
E (65594) vfs_native: Failed to initialize SDcard (263).
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
````

bdespatis, 2018-07-24:
I tested SDMODE_1Line and SDMODE_4Line, they work great. without pin pull up statement before.

Still wondering why SPI is no longer working since April Updates ...

luciengaitskell, 2018-08-05:
Any plan on addressing this in a future update? It's the main reason I haven't been able to implement this port, as I can't mount using an existing SD driver

luciengaitskell, 2018-08-05:
Just found this post on the forum: https://loboris.eu/forum/showthread.php?tid=216
I'll look out that commit in the future

loboris, 2018-08-28:
There were some updated to SDCard handling in the latest update.
`uos.sdconfig()` [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/filesystems#uossdconfigmode-clk-mosi-miso-cs-maxspeed-spihost) is updated.
You can try to set a lower spi speed if the default one does not work.
For example _Adafruit Feather wing_ requires the spi speed to be set to 16 MHz.