loboris/MicroPython_ESP32_psRAM_LoBo #194: CA, Cert, and Key

luke-unifymonitor, 2018-07-26:
Hi,

I'm following this article to allow for Just In Time Provisioning of my IOT Device/Thing:

https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core/

The call to mqtt requires a ca, cert, and key file to be supplied. 

$ mosquitto_pub --cafile root.cert --cert deviceCertAndCACert.crt --key deviceCert.key -h <prefix>.iot.us-east-1.amazonaws.com -p 8883 -q 1 -t foo/bar -I anyclientID --tls-version tlsv1.2 -m "Hello" -d

However, looking at https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/mqtt#create-mqtt-instance-object I can only see a single `cert	Certificate file name` property.

Am I able to point these extra credentials into MQTT Module?

Apologies if this is a silly question, I'm pretty new to SSL/X.509 etc..

Many thanks for the library and any help offered,

Luke.

loboris, 2018-08-16:
Sorry for the late reply, I was on a vacation.

Only cert is supported for now, the support for mutual authentication (cert+key) should be added in the next couple of weeks.

MagnusKriel, 2019-03-03:
Hi, has this been resolved? Or is it possible to port the MQTT C module of the AWS FreeRtos implementation?