loboris/MicroPython_ESP32_psRAM_LoBo #195: Where's flash.sh?

jrborbars, 2018-07-26:
Hi,
I'm downloaded the image from the link provided in the "prebuilt binaries". So, when I unzip them, there's no "flash.sh" in the directory. This seems necessary to write the binaries in the uC. 
Trying a workaround following an example on the internet, that shows a sequence of writes in the uC, initializing in 0x00001000, and next in 0x0000f000 and finally in 0x000b8000. Examining the length of the data (as a clue) I'm inferred that first is write the bootloader.bin file, the seconf is the phy_init_data.bin and last Micropython.bin. But without sucess. Can you, please, show the content of the flash.sh or republish again? Best regards

flashypepo, 2018-08-03:
The file flash.sh moved up one directory/folder. 
See the README in the github folder of the firmwares:
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/firmware

Was that your question?
Peter

jrborbars, 2018-08-29:
Yes, this answer my question. I'm following a tutorial when the location of flash.sh changed. After a while, I'm able to search and found the "missed" flash.sh. Thanks for your information.

closed 2018-08-29 by jrborbars