loboris/MicroPython_ESP32_psRAM_LoBo #196: stack overflow error

jarekdudzinski, 2018-07-28:
Encountered stack overflow with very simple program running. Screenshoots attached, feel free to ask anything - not sure what else should I attach.

PS: thanks for creating this micropython fork! it's really great :) 

[image]
[image]

loboris, 2018-08-16:
Thank you for reporting, I'll check if I can reproduce this.

loboris, 2018-08-28:
Could you, please, if you still have this issue after the latest update.