loboris/MicroPython_ESP32_psRAM_LoBo #200: readPixel not work for me

kcfkwok2003, 2018-08-13:
I have tried the tft module, the pixel(x,y,color) work fine. But when I call the readPixel(x,y) to read back the color, it always return 0xffffff. Have I made something wrong?

loboris, 2018-08-28:
Some displays may not support reading the data from graphics RAM (do not have MISO connected).

eflukx, 2018-08-28:
Even if reading is supported by the display, reading back from the external frame buffer probably is relatively slow anyway (i.e. _much_ slower then when using an in-RAM framebuffer).

loboris, 2018-08-28:
Reading is done via the same SPI interface as writting.
Maximum read speed is detected on initialization and is usually slower than the write speed (8-10 MHz).
The read speed is reported by the [`tft_setspeed(speed)` function](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/display#tfttft_setspeedspeed)

closed 2020-11-21 by kcfkwok2003