loboris/MicroPython_ESP32_psRAM_LoBo #202: OTA: Verify Firmware with signatures

kabel42, 2018-08-16:
To ensure that firmware transmitted to the device has not been tampered with, it would be good if we could check a signature instead of an md5 checksum.

The signature could either be provided from Python and checked in the updater or a hash of the firmware could be passed to Python code for check before switching the boot partition.

I have made an example of a modified ota_start, replacing md5 with sha256+ed25519, here: https://github.com/kabel42/MicroPython_ESP32_psRAM_LoBo/commit/4e5df694496cbdc3629f95e951aa44d1a4c2d970

Would you be interested in merging such a change? Should MD5 still be an option?

loboris, 2018-08-28:
Could you, please, explain in more details why do you think MD5 is not enough for file integrity check?

eflukx, 2018-08-28:
When using signatures, not only validity, but also the _origin_ of the firmware can be established with certainty

loboris, 2018-08-28:
Latest esp-idf versions have the componnent for OTA update over **https** where the server certificate can be provided to ensure the origin of the firmware.
I'm working to implement this in the MicroPython OTA module.
That should probably address your security concerns.

kabel42, 2018-08-28:
As @eflukx said, the idea is to keep an attacker from modifying code while it is being transmitted. https with certificate chain verification should be as good but might be more difficult to implement and set up.

loboris, 2018-09-01:
**ota** module is updated, **https** can now be used for firmware update and server **certificate** file can be used to verify the server.
[Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/ota#otastarturl--file-restart-md5-forcefactory-certificate)

kabel42, 2018-09-20:
Thanks, this works nicely

One more thing, if the Server is not reachable the code tries to free client twice and crashes.
This can be fixed by replacing `goto exit_client` with `goto exit` here: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/modota.c#L209

closed 2020-12-09 by kabel42