loboris/MicroPython_ESP32_psRAM_LoBo #204: Neopixel module doesn't work

mtiutiu, 2018-08-27:
I created a dumb example in order to test the Neopixel module:

```
import machine
import time

np = machine.Neopixel(22, 5, machine.Neopixel.TYPE_RGB)
np.brightness(100)

for pos in range(1, 5):
    np.set(pos, 0xFF0000)
    time.sleep_ms(100)
    np.show()
```

and the led strip doesn't light up at all. I checked the hardware connections and all that stuff and nothing wrong there. I also tested using the official micropython project neopixel module and that one works as expected so it doesn't seem to be a hardware issue to me.

loboris, 2018-08-27:
I have tested it and it works as expected.
Depending on your neopixel type, you may need to adjust the timings (WS2812 timings are default).

mtiutiu, 2018-08-27:
I'm using a standard ws2812 led strip. Sorry  I forgot to mention it. So it's strange that it works for you and not for me. But anyway I'll play a little bit with the timings setting and I'll come back with updates. Thanks for the info.

mtiutiu, 2018-08-28:
OK, I changed the timings using `np.timings(((600, 700), (800, 350), 60000))` and it started to light up but not as it should. When I want to set one led to red for example it lights up in another color (some kind of blueish). 

Another thing to mention is that it doesn't respect the brightness level when using for example `np.brightness(10, update=True)` - it stays at the max level all the time. I played with color order also in various combinations: BGR, RGB, etc. and the results are the same regarding color reproduction.

As a note I have a small led strip - around 5 leds which I use for testing and I'm using a level converter also but with or without it the results are still wrong.

I'm using a standard WS2812 led strip bought from Aliexpress some long time ago and it's not faulty because I used it with other libraries and it works. So I'm a little bit stuck here because I don't know what to try next...the only thing that's left for me to see is if the timings are generated correctly on an oscilloscope or on a logic level analyzer.

loboris, 2018-08-28:
This is strange, I've been using neopixel strips for a long time and never had a problem.
I always have neopixel strip's Vcc connected to 5V, data pin connected to the ESP32 pin (no level shifter).
I've been testing with small 8-leds strips and with 144-leds strips without problems.
Changing brightness works.

There were some reports that the timings for some neopixels must be reversed (H<->L values).
You can try to change the timings:
```
((600, 700), (350, 800), 60000)
((350, 800), (600, 700), 60000)
((700, 600), (800, 350), 60000)
((800, 350), (700, 600), 60000)
...
```

Here is the waveform generated for led1 set to RED & led2 set to BLUE, **default settings**:
[image]

Brightness changed to 128 (50%)
[image]

vrubel, 2018-08-28:
Boris, remember, I sent you an oscilograph video. It shows that periodically the number of pulses in the control sequence increases.

vrubel, 2018-08-28:
https://www.dropbox.com/s/nuub0xtzlkbtp5g/IMG_3193.MOV

loboris, 2018-08-28:
@vrubel
Thanks. I remember now. I was never able to reproduce this. I was using 144 leds strip playing effects for days without any issue.
When I find some time I'll check the Neopixel module once again more thoroughly.

mtiutiu, 2018-08-28:
@loboris Tried with your suggestions regarding timings but still no luck. And from time to time if I run that dumb sketch I get: `ValueError: Cannot acquire RMT channel`

loboris, 2018-08-28:
Regarding `ValueError: Cannot acquire RMT channel`, always use `np.deinit()` if you are creating the new Neopixel instance object on the same pin.

curlyz, 2020-05-22:
I am wondering why the Neopixel can't use pin 32, 33. 
```C
    if (pin > 31) {
    	mp_raise_ValueError("Wrong pin, only pins 0~31 allowed");
    }
```
Since they are also digital pin ? Should this be "pin > 33" ?