loboris/MicroPython_ESP32_psRAM_LoBo #209: E (1126) MicroPython: Error while opening MicroPython NVS name space

jasonblewis, 2018-09-08:
I'm trying to get the wifi working on this ESP32 board I have but i get this error:

MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32

at boot up I see:

```
I (759) cpu_start: Pro cpu start user code
D (771) clk: RTC_SLOW_CLK calibration value: 3472845
I (111) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
D (112) heap_init: New heap initialised at 0x3ffe0440
D (116) heap_init: New heap initialised at 0x3ffe4350
D (121) MicroPython: Entry
D (1124) MicroPython: SPIRAM: Disabled
D (1124) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (1125) MicroPython: Configure stack
E (1126) MicroPython: Error while opening MicroPython NVS name space
D (1133) MicroPython: MPy stack: 0x3ffc01ec - 0x3ffc51f4 (20488)
D (1139) MicroPython: Configure heap
D (1143) MicroPython: MPy heap: 0x3ffc51f8 - 0x3ffd9238 (81920)
D (1149) MicroPython: Main task exit, stack used: 1216
I (1151) MicroPython: [=== MicroPython FreeRTOS task started (sp=3ffc5140) ===]
```

then when I try and start networking:

```
>>> import network
>>> import machine
>>> import time
>>> 
>>> sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
W (367361) wifi: wifi osi_nvs_open fail ret=4353

E (367361) wifi: wifi_init_in_caller_task 707 ret=4353
E (367361) [modnetwork]: Error initializing WiFi (4353)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 4353
>>> 
```

I've tried flashing the arduino ESP32 wifiscan program and that successfully launches and scans for wifi networks and displays them on the terminal.

jasonblewis, 2018-09-08:
I tested with an older build and I don't get the error:

```
I (0) cpu_start: Starting scheduler on APP CPU.
D (99) heap_init: New heap initialised at 0x3ffe0440
D (103) heap_init: New heap initialised at 0x3ffe4350
D (108) MicroPython: Entry
D (1111) MicroPython: SPIRAM: Disabled
D (1111) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (1130) MicroPython: Configure stack
D (1130) nvs: nvs_open_from_partition MPY_NVM 1
D (1131) nvs: nvs_get MPY_StackSize 4
D (1131) nvs: nvs_get_str_or_blob MpyTimeZone
D (1136) MicroPython: MPy stack: 0x3ffc0c0c - 0x3ffc5c14 (20488)
D (1141) MicroPython: Configure heap
D (1145) nvs: nvs_get MPY_HeapSize 4
D (1149) MicroPython: MPy heap: 0x3ffc5c18 - 0x3ffd9c58
D (1154) MicroPython: Main task exit, stack used: 1392
I (1156) MicroPython: [=== MicroPython FreeRTOS task started (sp=3ffc5b60) ===]

W (1173) SPIFFS: mount failed, -10025. formatting...

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1048576; Flash address: 0x180000]
----------------
Filesystem size: 956416 B
           Used: 512 B
           Free: 955904 B
----------------
MicroPython ESP32_LoBo_v3.2.20 - 2018-06-28 on ESP32 board with ESP32
Type "help()" for more information.
>>> import network
>>> import machine
>>> import time
>>>
>>> sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
I (53802) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 0, 0
True
```
commit 210efe9a9e006938ce305ca69ee3da9096c09330

loboris, 2018-09-08:
I couldn'r reproduce this.

Could you try to erase the flash (`./BUILD.sh erase`) and try again.

jasonblewis, 2018-09-08:
erased then flashed from the source build and was able to connect to wifi
```
>>> ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4844
load:0x40078000,len:8652
ho 0 tail 12 room 4
load:0x40080000,len:5748
entry 0x400802dc

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 2686976; Flash address: 0x170000]
----------------
Filesystem size: 2462464 B
           Used: 512 B
           Free: 2461952 B
----------------
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>
>>> sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
True
>>> sta_if.connect(wifi_ssid, wifi_passwd)
>>> 
>>> time.sleep(6)
>>> sta_if.ifconfig()
('192.168.1.237', '255.255.255.0', '192.168.1.1', '192.168.1.1')
>>> 
```
and also I tested with the prebuilt firmware from github and that seems to work fine too now. So I'm not sure why I was having the issue before, or maybe the `./BUILD.sh erase` fixed it.

thanks.

closed 2018-09-08 by jasonblewis