loboris/MicroPython_ESP32_psRAM_LoBo #210: Interrupt callback not scheduled during utime.sleep?

ubIQio, 2018-09-09:
I have a 10Hz pulse stream on pin4 .

When I paste the following code  into the REPL

```
import machine
interruptCounter = 0

def callback(pin):
  global interruptCounter
  interruptCounter = interruptCounter+1

p4 = machine.Pin(4, machine.Pin.IN, machine.Pin.PULL_UP)

p4.irq(trigger=machine.Pin.IRQ_FALLING, handler=callback)

```

followed by
`print (interruptCounter)`

as expected I get a number that increments in the background at ~10Hz, viewable at any subsequent print.

However, when I put the same basic code in Main.py, the increment only happens at the loop rate, ie 1Hz.

```
import utime

import machine
interruptCounter = 0

def callback(pin):
  global interruptCounter
  interruptCounter = interruptCounter+1

p4 = machine.Pin(4, machine.Pin.IN, machine.Pin.PULL_UP)

p4.irq(trigger=machine.Pin.IRQ_FALLING, handler=callback)

while True:
    print (interruptCounter)
    utime.sleep_ms(1000)

```
It appears that the callback does not get scheduled during utime.sleep_ms()
Tried using _thread.wait() with same result.

Any ideas how to fix?

loboris, 2018-09-09:
Pin interrupt callbacks **are** scheduled during `utime.sleep_ms()`, but **beeing scheduled** means the callback function will be executed as soon as possible, which is **after** the `utime.sleep_ms()` finishes.
No **Python** function can be executed while some built-in C-function runs.

You can change your while loop to detect the Pin change count (with 10ms resolution):
```python
def test():
    lastCounter = interruptCounter
    while True:
        if lastCounter != interruptCounter:
            lastCounter = interruptCounter
            print (interruptCounter)
        utime.sleep_ms(10)

test()
```

You can use `_thread.wait()` (inside the thread function) which **will be** interrupted, but you have to send a **thread notification** from from Pin interrupt callback.

ubIQio, 2018-09-09:
Thanks Lobo, not sure I understand re threading...

If the callback can send the thread notification during a _thread.wait() in  the MainThread loop, then the interruptCounter in my example will be incremented, so should be seen by the MainThread?

But it isnt...

loboris, 2018-09-09:
**main** thread (REPL) **must not** block (with `utime.sleep_ms`) for Pin interrupt callback function to be executed.
In fact, no **Python** function should block any longer than absolutely necessary if you want some **Python** background processing.
You can define a Python function for sleep which will allow callbacks and threads to be executed:
```python
def py_sleep_ms(tm):
    tmo = tm
    while tmo > 0:
        utime.sleep_ms(3)
        tmo -= 3
```
The walue or 3 in `utime.sleep_ms` is used because `utime.sleep_ms(ms)` does not block only for `ms>2`.

**Implementing `utime.sleep_ms` and `utime.sleep` functions with an optional argument which would allow to _break_ if any scheduled function is waiting could be implemented.** I'll test it.
In that case `utime.sleep_ms` could return the number of ms passed, so you could continue the sleep in case it was interrupted by breaking on scheduled callback.

I your case, you can simply put `print (interruptCounter)` in the callback function.

ubIQio, 2018-09-10:
Thanks Boris, that will be perfect as I'm actually trying to count pulses
as short as  2ms so cannot poll at 3ms intervals...

This would essentially interrupt any wait loop, so would be great.

Another way would be to use the ULP as pulse counter, are you  planning
support for that cf  Nils' fork of your repo?

https://github.com/NilsNoreyson/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/main

--

Jonathan Peace

ubIQio  *- Smart IO, everywhere <https://ubiqio.com>*