loboris/MicroPython_ESP32_psRAM_LoBo #216: CAN bus implementation

ldecicco, 2018-09-18:
I have come across a can bus implementation for esp-idf (https://github.com/ThomasBarth/ESP32-CAN-Driver). There is also an implementation for the pyboard available here https://github.com/micropython/micropython/blob/master/tests/pyb/can.py.

Are there any plans to support CAN BUS?

Thanks!

alankaranak, 2018-09-24:
I join the issue! Plan to work with [it](https://www.tindie.com/products/Fusion/can32-an-esp32-dev-board-with-can-bus-v2/)

loboris, 2018-10-05:
CAN support will be implemented in a couple of months.
I've just got some hardware to test it, but I'm very busy with some other tasks, I'll try to work on it as soon as possible.

wangshujun-tj, 2019-02-14:
Looking forward to CAN support on esp32, I relied too much on CAN BUS before.

Civilduino, 2020-03-13:
Hi Boris, was this ever implemented?