loboris/MicroPython_ESP32_psRAM_LoBo #218: 'CPU halted' reported when ruining ping-pong using _thread module.

KKawase0104, 2018-09-20:
Hi,

When running ping-pong program using serverTest.py and clientTest.py, 'CPU halted.' was reported on the server side. I checked heap size left each time the server received client's message and found heap memory wasn't release.

> serverTest.py
```
import network
import socket
import _thread
import gc

def server_process():
  ap = network.WLAN(network.AP_IF)
  ap.active(True)
  ap.config(essid='esp32_ap')
  ap.config(authmode=3, password='esp32esp32')
  ip = ap.ifconfig()
  
  s = socket.socket()
  s.bind((ip[0], 50007))
  s.listen(1)
  while True:
    csock, caddr = s.accept()
    print(csock)
    print(caddr)
    _thread.start_new_thread("THRD#1", process, (csock, caddr))
    #_thread.start_new_thread(process, (csock, caddr)) # Original MP doesn't occur "CPU halted."
    #process(csock, caddr) # single thread doesn't occur "CPU halted."

def process(s, addr):
  print("server process start")
  i = 0
  while True:
    i += 1
    rcvmsg = s.recv(1024)
    print('Received -> %s : %d, %d' % (rcvmsg, i, gc.mem_free()))
    s.send("received")

  s.close()

server_process()
```

> clientTest.py

```
import network
import socket
import _thread
import time

st = network.WLAN(network.STA_IF)
st.active(True)
st.connect('esp32_ap', 'esp32esp32')

while not st.isconnected():
  pass
  
print("conneted!!")

ip = st.ifconfig()

s = socket.socket()
s.connect((ip[3], 50007))
while True:
  sent = s.send("from " + ip[0])
  response = s.recv(4096)
  print (response)
```

I tried followings:
prepared tow ESP32 boards (ESP32_s: server, ESP32_c:client) , uploaded serverTest.py to ESP32_s and clientTest.py to ESP32_c and run command using REPL on each board (as follows).

> ESP32_s's REPL
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> import serverTest
```
> ESP32_c's REPL
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> import clientTest
```

thanks.

loboris, 2018-10-05:
I've tested with your code and it runs without error.
I'll do some more test with similar setup in a couple of days when I'l have some more time.

KKawase0104, 2018-10-09:
Thank you for your reply.
As I tried three firmwares (esp32, esp32_all, esp32_ota), 'CPU halted' reported.
The log is following.

```
>>> import server
W (5403) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (5552) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 2
<socket>
('192.168.4.2', 58565)
server process start
Received -> b'from 192.168.4.2' : 1, 72352
Received -> b'from 192.168.4.2' : 2, 72208
Received -> b'from 192.168.4.2' : 3, 72064
Received -> b'from 192.168.4.2' : 4, 71920
...
Received -> b'from 192.168.4.2' : 491, 1792
Received -> b'from 192.168.4.2' : 492, 1648
Received -> b'from 192.168.4.2' : 493, 1504
Received -> b'from 192.168.4.2' : 494, 1360
Received -> b'from 192.168.4.2' : 495, 1216
Received -> b'from 192.168.4.2' : 496, 1072
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401cd2eb  PS      : 0x00060e30  A0      : 0x800fc59b  A1      : 0x3ffc4bc0
A2      : 0x00000001  A3      : 0x3f414b84  A4      : 0x3ffc7220  A5      : 0x3ffc4c80
A6      : 0x3ffc4c70  A7      : 0x3ffc4bd0  A8      : 0x3f41755c  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000019  A12     : 0x3ffc5a00  A13     : 0x3ffc4990
A14     : 0x3f416f14  A15     : 0x3ffe5020  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x401cd2eb:0x3ffc4bc0 0x400fc598:0x3ffc4be0 0x400f346e:0x3ffc4c80 0x400eeab5:0x3ffc4ce0 0x400eeae2:0x3ffc4d00 0x400ef63f:0x3ffc4d20 0x400f8ebf:0x3ffc4db0 0x400f8f0f:0x3ffc4dd0 0x400f91b2:0x3ffc4e00 0x400ef4bb:0x3ffc4ed0 0x400fc2f0:0x3ffc4f10 0x400f346e:0x3ffc4fb0 0x400eeab5:0x3ffc5010 0x400eeae2:0x3ffc5030 0x400e164f:0x3ffc5050 0x400e18cd:0x3ffc5100 0x400d545c:0x3ffc5140

CPU halted.
```

It may not be happen on the spram version of the firmware.
I can't confirm it because I don't have a ESP32 board with SPIRAM.
I had to write the firmware version, sorry.

If the problem is specific to a firmware that isn't a spram version,
is there a solution?

Thank you.

KKawase0104, 2018-10-25:
Hello.

Compared with the original MicroPython, I found that the CLEAR_ON_SWEEP definition in "MicroPython_BUILD/components/micropython/py/gc.c" is different.

When I defined this definition as 0 just as original is, 'CPU halted' no longer occurs.

With the original MicroPython, even if I defined CLEAR_ON_SWEEP as 1, it runs without 'CPU halted'.
I think this means something.

I'm still tackling this 'CPU halted' situation.
For me, MicroPython_ESP32_psRAM_LoBo is very good to use and I'd really like to solve this problem by all means.

Please let me know if you notice something.

Thank you.