loboris/MicroPython_ESP32_psRAM_LoBo #219: Create mqtt instance object fails

Pjort, 2018-09-24:
From the wiki it seems I only need to import the network lib and the create the mqtt object.

Login succeeded!
Type "help()" for more information.

`>>> import network
>>> mqtt = network.mqtt('local', 'mqtt://192.168.2.8', user='foo', password='bar', cleansession=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'mqtt'
>>>`

But that seems to fail for me.

Pjort, 2018-09-25:
No luck

```
>>> import network
>>> client = network.mqtt('local', 'mqtt://192.168.2.8', user='foo', password='bar', cleansession=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'mqtt'
```

flashypepo, 2018-09-25:
Ha, now it gets tough ;-)

Reading my code, I also see that the MQTT-server address in the connect-call is plain IP-address, not prefixed with 'mqtt:'. It might be a reason to try removing it from the MQTT-server address.

Here is my complete MQTT-client code, running on an ESP32, which publish temperature and humidity to a MQTT-broker, which runs on a Raspbery Zero. ESP32 and Raspberry are on local network and it works!

Remarks:
* module dhtsensor is my own module; its a class wrapped around machine.DHT.
* I  based my code on that of a blog: https://www.rototron.info/raspberry-pi-esp32-micropython-mqtt-dht22-tutorial/, and adopted it to Loboris micropython, also  using the examples/wiki of Loboris (github source code/example).

--------- start of code
# MQTT client that publishes values about topic 'temp_humidity'
# 2018-0923 new, Loboris Micropython

import network
from time import sleep
from machine import Pin, DHT
import dhtsensor
import led

MQTT_BROKER = "192.168.178.29"  # MQTT Server Address (Change to the IP address of your Pi)
CLIENT_ID = "LOLIND32_PRO"  #2018-0923: this name must not be too long, strange!
TOPIC = b'temp_humidity'

# Loboris MQTTClient with callbacks
def conncb(task):
    print("[{}] Connected".format(task))

def disconncb(task):
    print("[{}] Disconnected".format(task))

def subscb(task):
    print("[{}] Subscribed".format(task))

def pubcb(pub):
    print("[{}] Published: {}".format(pub[0], pub[1]))

def datacb(msg):
    print("[{}] Data arrived from topic: {}, Message:\n".format(msg[0], msg[1]), msg[2])

client = network.mqtt(CLIENT_ID, MQTT_BROKER,
    cleansession=True,
    connected_cb=conncb,
    disconnected_cb=disconncb,
    subscribed_cb=subscb,
    published_cb=pubcb,
    data_cb=datacb,
)
sleep(1) #wait some time for connection, it works but needed??
status = client.status()[1]
print('MQTT status:', status)
client.start()

# create sensor type DHT22 at GPIO15
sensor = dhtsensor.DHTSensor(15, DHT.DHT2X)
print('DHT type:', sensor.type)
print('Sensor:', sensor.sensor)
print('Sensor-pin:', sensor.pin)

led = led.Led(5) #signal MQTT transfer
led.on() # builtin-led off!

# get sensor values and publish
while True:
    try:
        led.off() # builtin-led on: measurement
        t, h = sensor.read()
        #OK, DEBUG: print('Temperature = {0:3.1f} Celsius, Humidity = {1:3.1f} %'.format(t, h))
        if isinstance(t, float) and isinstance(h, float):  # Confirm sensor results are numeric
            msg = (b'{0:3.1f},{1:3.1f}'.format(t, h))
            client.publish(TOPIC, msg) # Loboris
            #debug: print(msg)
        else:
            print('Invalid sensor readings.')
    except OSError:
        print('Failed to read sensor.')

    led.on() # builtin-led off: silence
    sleep(4)
---- end of code

Good luck, let us know your results,
regards, Peter

flashypepo, 2018-09-25:
OMG, sorry for the bad layout! 
Hopefully you understand whats going on.
Peter

mtiutiu, 2018-09-25:
@Pjort 

I tried your snippet on my esp32 board using the latest precompiled firmware and it doesn't throw any exceptions/errors or whatsoever. I'm using atom and pymakr package for doing development which is a great match btw. I ran your two lines of code inside the atom's pymakr console and it worked like a charm. 

I see there a line above your code "Login succeded!" - are you connecting to the micropython console via telnet?

Pjort, 2018-09-25:
> @Pjort
> 
> I tried your snippet on my esp32 board using the latest precompiled firmware and it doesn't throw any exceptions/errors or whatsoever. I'm using atom and pymakr package for doing development which is a great match btw. I ran your two lines of code inside the atom's pymakr console and it worked like a charm.
> 
> I see there a line above your code "Login succeded!" - are you connecting to the micropython console via telnet? It seems so...maybe there's something related to that. But from a standard serial console it works as expected.
> 
> Still, I tried that just now and it worked from telnet also without any exception or errors.

I compiled it my self, where is the precompiled version?

Yes I am using a telnet connection. But I just did that to ilustrate the problem, I couldn't get it to work by running a script either.

But it might be my compiled version that is bad, where is the precompiled?

mtiutiu, 2018-09-26:
@Pjort 

Sorry for the late response. [Here](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares#download-links-to-the-latest-firmwares) it is.

Pjort, 2018-09-26:
> @Pjort
> 
> Sorry for the late response. [Here](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares#download-links-to-the-latest-firmwares) it is. If it's customized and compiled by you then maybe you forgot to add some module(s). Other than that I don't know what could cause the issue. I always used the prebuilt firmware (the **ALL** variant) and I didn't had any issues so far.
> 
> Please let us know how it goes.

Haha, that did the trick. Usually I find that prebuilds are found in the release tab in github. That is why I assumed there wasn't any prebuilds, and I had to build it my self. Thanks a lot. Now it works.

closed 2018-09-26 by Pjort