loboris/MicroPython_ESP32_psRAM_LoBo #221: odroid go support?!?

jedie, 2018-10-02:
I have a ODROID-GO and whant to use microPython... 

Specs from https://github.com/hardkernel/ODROID-GO-MicroPython :

- Custom ESP32-WROVER(16 MB Flash Memory)
- 4MB PSRAM

So i guess **esp32_psram_all** is the best choice?!?

I made this:

`MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/firmware/esp32_psram_all$ ../flash.sh`

And the REPL works. But i get errors like:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

The boot output is:

```
ESP32: soft reboot
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5976
load:0x40078000,len:9152
ho 0 tail 12 room 4
load:0x40080000,len:6120
entry 0x40080334
[0;32mI (30) boot: ESP-IDF v3.1-rc1-53509c7b 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 13:16:56[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 16MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (74) boot:  2 MicroPython      factory app      00 00 00010000 00220000[0m
[0;32mI (82) boot:  3 internalfs       Unknown data     01 82 00230000 00100000[0m
[0;32mI (89) boot: End of partition table[0m
[0;32mI (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x77898 (489624) map[0m
[0;32mI (274) esp_image: segment 1: paddr=0x000878c0 vaddr=0x3ffb0000 size=0x0574c ( 22348) load[0m
[0;32mI (283) esp_image: segment 2: paddr=0x0008d014 vaddr=0x3ffb574c size=0x00000 (     0) load[0m
[0;32mI (283) esp_image: segment 3: paddr=0x0008d01c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (292) esp_image: segment 4: paddr=0x0008d424 vaddr=0x40080400 size=0x02bec ( 11244) load[0m
[0;32mI (305) esp_image: segment 5: paddr=0x00090018 vaddr=0x400d0018 size=0x175798 (1529752) map[0m
[0;32mI (845) esp_image: segment 6: paddr=0x002057b8 vaddr=0x40082fec size=0x1a350 (107344) load[0m
[0;32mI (890) esp_image: segment 7: paddr=0x0021fb10 vaddr=0x400c0000 size=0x00714 (  1812) load[0m
[0;32mI (891) esp_image: segment 8: paddr=0x0022022c vaddr=0x50000000 size=0x0098c (  2444) load[0m
[0;32mI (916) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (916) boot: Disabling RNG early entropy source...[0m
[0;32mI (918) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (922) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (929) cpu_start: Pro cpu up.[0m
[0;32mI (933) cpu_start: Starting app cpu, entry point is 0x400816dc[0m
[0;32mI (924) cpu_start: App cpu up.[0m
[0;32mI (1832) spiram: SPI SRAM memory test OK[0m
[0;32mI (1833) heap_init: Initializing. RAM available for dynamic allocation:[0m
D (1833) heap_init: New heap initialised at 0x3ffae6e0[0m
[0;32mI (1838) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
D (1844) heap_init: New heap initialised at 0x3ffbdb28[0m
[0;32mI (1849) heap_init: At 3FFBDB28 len 000224D8 (137 KiB): DRAM[0m
[0;32mI (1856) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1862) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
D (1869) heap_init: New heap initialised at 0x4009d33c[0m
[0;32mI (1874) heap_init: At 4009D33C len 00002CC4 (11 KiB): IRAM[0m
[0;32mI (1880) cpu_start: Pro cpu start user code[0m
[0;32mI (1885) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
D (1900) clk: RTC_SLOW_CLK calibration value: 3248832[0m
[0;32mI (122) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
D (123) heap_init: New heap initialised at 0x3ffe0440[0m
D (127) heap_init: New heap initialised at 0x3ffe4350[0m
D (132) MicroPython: Entry[0m
D (1135) MicroPython: SPIRAM: Enabled[0m
D (1145) nvs: nvs_flash_init_custom partition=nvs start=9 count=6[0m
D (1147) MicroPython: Configure stack[0m
D (1147) nvs: nvs_open_from_partition MPY_NVM 1[0m
D (1149) nvs: nvs_get MPY_StackSize 4[0m
D (1152) nvs: nvs_get_str_or_blob MpyTimeZone[0m
D (1157) MicroPython: MPy stack: 0x3ffc126c - 0x3ffc6274 (20488)[0m
D (1163) MicroPython: Configure heap[0m
D (1166) nvs: nvs_get MPY_HeapSize 4[0m
D (1170) MicroPython: MPy heap: 0x3f80001c - 0x3fb0005c (3145728)[0m
D (1176) MicroPython: Main task exit, stack used: 1392[0m
[0;32mI (1185) MicroPython: [=== MicroPython FreeRTOS task started (sp=3ffc61c0) ===]
[0m

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1048576; Flash address: 0x230000]
----------------
Filesystem size: 956416 B
           Used: 512 B
           Free: 955904 B
----------------
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> 
```

loboris, 2018-10-05:
When exactly do you get the error
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```
? Just waiting in the REPL, or executing some command?

jedie, 2018-10-06:
e.g.: sending a file with thonny results in:
```
>>> %upload main.py /main.py
Traceback (most recent call last):
  File "/home/jens/repos/thonny-Env/src/thonny/thonny/plugins/micropython/__init__.py", line 860, in _check_and_upload
    self._upload(source, target)
  File "/home/jens/repos/thonny-Env/src/thonny/thonny/plugins/micropython/__init__.py", line 867, in _upload
    self._upload_via_serial(source, target)
  File "/home/jens/repos/thonny-Env/src/thonny/thonny/plugins/micropython/__init__.py", line 878, in _upload_via_serial
    self._execute_and_expect_empty_response("__upf = open(%r, 'wb')" % target)
  File "/home/jens/repos/thonny-Env/src/thonny/thonny/plugins/micropython/__init__.py", line 503, in _execute_and_expect_empty_response
    assert len(err) == 0, "Error was " + repr(err)
AssertionError: Error was bytearray(b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nOSError: [Errno 19] ENODEV\r\n')
>>> 
```

But something like this works:
```
>>> import os
>>> os.listdir()
['boot.py']
>>> open("boot.py", "r").read()
"# This file is executed on every boot (including wake-boot from deepsleep)\nimport sys\nsys.path[1] = '/flash/lib'\n"
>>> open("main.py", "w").write("#test")
5
>>> open("main.py", "r").read()
'#test'
```

I also try to send files via rshell doesn't work:

```
+ rshell -p /dev/ttyUSB0 rsync -v . /pyboard
Connecting to /dev/ttyUSB0 ...
Adding /pyboard/crossbar_demo.py
timed out or error in transfer to remote
```

If i add --debug i can see that the timeout appears in:
```
----- About to send 2174 bytes of code to the pyboard -----
def recv_file_from_host(src_file, dst_filename, filesize, dst_mode='wb'):
...
```

How can I better find out where the problem appears?

loboris, 2018-10-06:
Sorry, I've never used thonny, rshell was reported to work, but I'm not using it.
I can only recommend to use [FTPServer](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/ftpserver) for file transfer.

jedie, 2018-10-06:
Great idea. I will test it. But it would be interesting what happens in thonny. I have use ist with baseline micropython

jedie, 2018-10-08:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

Comes from: `open("/foo_bar.py", "wb")`
It will work, if i change the file path from '/foo_bar.py' to 'foo_bar.py'

jedie, 2018-10-09:
OK... Then this can be closed... But: Is it possible to change the `OSError: [Errno 19] ENODEV` message to something more appropriate? e.g.: `OSError: [Errno 19] Target not writable!`

loboris, 2018-10-09:
`ENODEV` is a standard error code (OS error) which means "No such device" and will be issued if some requested _device_ (which may be file, but also something else) cannot be accessed.
It would be impractical to generate more descriptive messages for all possible cases.

jedie, 2018-10-09:
Ah ok! Good to know...

But can be `ENODEV` renamed to `No such device` ?