loboris/MicroPython_ESP32_psRAM_LoBo #222: CPU halted when using MicroDNSSrv

dukeduck1984, 2018-10-04:
Hi,

When I was using the MicroDNSSrv, the following error occured:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400feb13  PS      : 0x00060030  A0      : 0x800f58f3  A1      : 0x3ffec020  
A2      : 0x00000000  A3      : 0x3f817220  A4      : 0x3f813330  A5      : 0x00000000  
A6      : 0xffffff30  A7      : 0x3f81ca04  A8      : 0x800fea74  A9      : 0x3ffec000  
A10     : 0x00000000  A11     : 0x3f813320  A12     : 0x3f418554  A13     : 0x3f418b64  
A14     : 0x3f418b64  A15     : 0x3ffec0b0  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4009e268  LEND    : 0x4009e296  LCOUNT  : 0xffffffff  

Backtrace: 0x400feb13:0x3ffec020 0x400f58f0:0x3ffec0c0 0x400f1231:0x3ffec0f0 0x400f3da0:0x3ffec110 0x400f3dbd:0x3ffec160 0x400f1231:0x3ffec180 0x400fcecc:0x3ffec1a0 0x400da019:0x3ffec240

CPU halted.

the code in main.py as following
```python
import network
from microDNSSrv import MicroDNSSrv

ap = network.WLAN(network.AP_IF) 
ap.active(True) 
ap.config(essid='test_ESP32') 
host_ip_addr = ap.ifconfig()[0] 

if MicroDNSSrv.Create({
    'test-domain.local': host_ip_addr,
    '*': host_ip_addr,}):
    print('MicroDNSSrv started.')
else:
    print('Error to starts MicroDNSSrv...')
```

Thanks for the help.

-Kaiyuan

loboris, 2018-10-05:
I haven't tested MicroDNSSrv, so I can't help at the moment.

Have you tried the built-in [mDNS Module](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/mdns) ?

dukeduck1984, 2018-10-05:
> 
> 
> I haven't tested MicroDNSSrv, so I can't help at the moment.
> 
> Have you tried the built-in [mDNS Module](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/mdns) ?

Hi Boris,

Thanks for the reply.

Yes, I have tried the mDNS but didn't get it to work.  My intention was to use the cell phone to visit a webpage in which the code controls the GPIOs of the ESP32 and the webpage was served by the ESP32 - I start the WLAN in AP mode,  and once the phone connects to the WIFI of ESP32, I wish to visit that webpage by a domain name rather than ESP32's IP adress.

I tried the mDNS as following:
```python
import network

dns = network.mDNS()
dns.start("testESP","MicroPython with mDNS")
_ = mdns.addService('_http', '_tcp', 80, "MicroPython", {"board": "ESP32", "service": "testEPS Web server"})
```

I assume I should be able to visit the webpage by http://testESP.local while the phone is connecting to the WIFI of ESP32?  But it didn't work out -  I had to type in the IP address of the ESP32.

Thanks for the help.
- Kaiyuan

loboris, 2018-10-05:
I have no issues with mDNS and AP mode (mDNS initialized the same way as in your example):
Tested on Ubuntu 18.04

**avahi**:
```
boris@UbuntuMate:~$ avahi-browse -avtr
Inačica poslužitelja: avahi 0.7; Naziv poslužitelja: UbuntuMate.local
E Ifce Prot Naziv                                         Vrsta                domena
...
...
= wlx180f7602ea61 IPv4 MicroPython                                   Web Site             local
   hostname = [testESP.local]
   address = [192.168.4.1]
   port = [80]
   txt = ["service=mPy Web server" "board=ESP32"]
```
**ping**:
```
boris@UbuntuMate:~$ ping testESP.local
PING testESP.local (192.168.4.1) 56(84) bytes of data.
64 bytes from 192.168.4.1 (192.168.4.1): icmp_seq=1 ttl=255 time=5.28 ms
64 bytes from 192.168.4.1 (192.168.4.1): icmp_seq=2 ttl=255 time=1.43 ms
64 bytes from 192.168.4.1 (192.168.4.1): icmp_seq=3 ttl=255 time=1.39 ms
64 bytes from 192.168.4.1 (192.168.4.1): icmp_seq=4 ttl=255 time=4.28 ms
64 bytes from 192.168.4.1 (192.168.4.1): icmp_seq=5 ttl=255 time=1.07 ms
^C
--- testESP.local ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4005ms
rtt min/avg/max/mdev = 1.075/2.695/5.288/1.742 ms
boris@UbuntuMate:~$ 
```
**Browser**:
[image]

dukeduck1984, 2018-10-07:
Thanks for trying Boris.

Strange.  I tried again, no luck.  Both Firefox in Win10 and my Android phone can't find the hosename, but no problem visiting the webpage by it's IP address.  BTW, I used Picoweb and the http server runs on port 80.  I also have issue with MicroWebSrv - the server is up and running, but I always got the 404 error no matter how I tried.  The similar issue is mentioned in this blog as well:  https://www.cnx-software.com/2017/10/16/esp32-micropython-tutorials/

-Kaiyuan

loboris, 2018-10-07:
The client OS must support [mDNS](https://en.wikipedia.org/wiki/Multicast_DNS).
It looks there are some issues on Win10 (see [this](https://superuser.com/questions/1330027/how-to-enable-mdns-on-windows-10-build-17134)).
There may be similar [issues](https://android.stackexchange.com/questions/49188/how-to-get-mdns-working-for-chrome-on-android) on Android. Try to install [Service browser](https://play.google.com/store/apps/details?id=com.druk.servicebrowser) or similar app to see if the ESP32 mDNS name is visible.

dukeduck1984, 2018-10-07:
Ah!  Thanks for pointing this out - very helpful!!

Kaiyuan

closed 2018-10-07 by dukeduck1984