loboris/MicroPython_ESP32_psRAM_LoBo #224: pin state value reported incorrectly

carterw, 2018-10-04:
This seems really basic, and it appears to be broken. I have an LED attached to VCC through a resistor and then to pin 2. I can toggle it on and off, but the reported pin state is always zero.

>>> from machine import Pin
>>> pObj = Pin(2,  Pin.OUT, value=1)   # LED turns off
>>> pObj.value()
0
>>> pObj.value(0)   # LED turns on
>>> pObj.value()
0

loboris, 2018-10-05:
As mentioned in the [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/pin#pinvalueval)

> If the pin is set to output-only mode (OUT or OUT_OD), the returned value will always be 0
> If you want to get the value of the output pin, it must be se to INOUT or INOUT_OD mode.

So, you should use `pObj = Pin(2,  Pin.INOUT, value=1)` .
It is rarely necessary, as you usually know to what state you've set the pin.

carterw, 2018-10-05:
Thanks. Can you look at the de-bounce issue?

closed 2018-10-05 by carterw

capedra, 2018-10-11:
> Thanks. Can you look at the de-bounce issue?

@loboris Hello, I dealt with this de-bounce issue using the following link. The issue is that the following script only works with the original firmware from micropython.org because **irq** function doesn't exist anymore (**irqvalue** doesn't work with it):
https://gist.github.com/SpotlightKid/0a0aac56606286a80f860b116423c94f

How may we deal with it? Thanks!