loboris/MicroPython_ESP32_psRAM_LoBo #225: network.LAN module static

elektrobil, 2018-10-05:


loboris, 2018-10-05:
I cannot reproduce this, assigning static IP works without issues.

```
import network, requests

>>> l=network.LAN(mdc=23,mdio=18, power=5)
>>> l.ifconfig()
('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')
>>> l.ifconfig(("192.168.0.46", "255.255.255.0", "192.168.0.1", "8.8.8.8"))
>>> l.ifconfig()
('192.168.0.46', '255.255.255.0', '192.168.0.1', '8.8.8.8')
>>> l.isconnected()
False
>>> l.active(True)
True
>>> l.isconnected()                                                                                                                         
True                                                                                                                                        
>>> l.ifconfig()                                                                                                                            
('192.168.0.46', '255.255.255.0', '192.168.0.1', '8.8.8.8')                                                                                 
>>> 
>>> requests.get("http://loboris.eu/ESP32/info.txt")                                                                                        
(200, 'Date: Fri, 05 Oct 2018 13:05:51 GMT\r\nServer: Apache/2.4.29 (Ubuntu)\r\nLast-Modified: Thu, 05 Oct 2017 11:08:15 GMT\r\nETag: "181-5
5acabda6ed78"\r\nAccept-Ranges: bytes\r\nContent-Length: 385\r\nVary: Accept-Encoding\r\nContent-Type: text/plain\r\n', '================\nW
elcome to ESP32\n================\n\nThe ESP32 is a low cost, low power microcontroller\nwith integrated Wi-Fi & dual-mode Bluetooth,\nwhich
 employs a dual-core\nTensilica Xtensa LX6 microprocessor.\n\nESP32 is created and developed by Espressif Systems,\na Shanghai-based Chinese
 company,\nand is manufactured by TSMC using their 40 nm process.\n\n---------\n2017 LoBo\n---------\n')                                    
>>> 
```

Ping from PC:
```
boris@UbuntuMate:~$ ping 192.168.0.46
PING 192.168.0.46 (192.168.0.46) 56(84) bytes of data.
64 bytes from 192.168.0.46: icmp_seq=1 ttl=255 time=0.412 ms
64 bytes from 192.168.0.46: icmp_seq=2 ttl=255 time=0.244 ms
64 bytes from 192.168.0.46: icmp_seq=3 ttl=255 time=0.250 ms
64 bytes from 192.168.0.46: icmp_seq=4 ttl=255 time=0.208 ms
^C
--- 192.168.0.46 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3035ms
rtt min/avg/max/mdev = 0.208/0.278/0.412/0.080 ms
boris@UbuntuMate:~$ 
```

elektrobil, 2018-10-05:
Sorry Boris

I mixed the subnet and gateway in my code. It's. working good sorry about that

Best regards.

closed 2018-10-05 by elektrobil