loboris/MicroPython_ESP32_psRAM_LoBo #226: GSM and GPS using together

neoleesky, 2018-10-06:
Hi lobo,
I'm using a SIM7600 module, it has GPS, GSM function. I can only use the  serial port of SIM7600 for GSM.
Is there any way to use GPS function of SIM7600 by serial port?

loboris, 2018-10-06:
AT commands can be used to start and access GPS.
I have one SIM7000E module which also have GPS.
I'm going to test it next week and I'll probably implement some GSM related methods in the MicroPython's GSM module.

neoleesky, 2018-10-06:
> AT commands can be used to start and access GPS.
> I have one SIM7000E module which also have GPS.
> I'm going to test it next week and I'll probably implement some GSM related methods in the MicroPython's GSM module.

YES, AT commands can access GPS

> >>> gsm.atcmd('AT+CGPSINFO',printable=True)                                                           
> '..+CGPSINFO: ,,,,,,,,....OK..' 

waiting your GPS implement

Thanks

closed 2018-10-07 by neoleesky