loboris/MicroPython_ESP32_psRAM_LoBo #227: Noob Question: How to run the examples?

superquest, 2018-10-08:
I'm a Python developer brand new to MicroPython . How can I run the examples?

Sorry for such a basic question but I don't see instructions anywhere and don't know enough MicroPython to figure it out ...

superquest, 2018-10-08:
I may have a problem flashing the firmware. I run `./BUILD.sh` and `./BUILD.sh flash` and encounter no errors. But I'm unable to connect to the device and get a micropython prompt using picotool. 

I am, however, able to get a prompt using picotool after following [this tutorial to install the generic esp32 firmware](https://appelsiini.net/2018/m5stack-esp32-firmware-cli/).

superquest, 2018-10-08:
By the way I'm using arch linux.

I added my user to the `uucp` group that owns `/dev/ttyUSB0`.

Now I get a silent failure:
```
$ ./BUILD.sh flash

---------------------
MicroPython for ESP32
---------------------

=========================================
Flashing MicroPython firmware to ESP32...
=========================================
Building partitions from /home/justin/dev/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
AR build/libnmea/liblibnmea.a
LD build/MicroPython.elf
esptool.py v2.5.0
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000 )...
esptool.py v2.5.0
Serial port /dev/ttyUSB0
'make flash' FAILED!
```

superquest, 2018-10-08:
monitor command also failing ...

```
$ ./BUILD.sh monitor

---------------------
MicroPython for ESP32
---------------------

============================
Executing esp-idf monitor...
============================
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

Traceback (most recent call last):
  File "/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/idf_monitor.py", line 734, in <module>
    main()
  File "/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/idf_monitor.py", line 659, in main
    monitor.main_loop()
  File "/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/idf_monitor.py", line 333, in main_loop
    self.handle_serial_input(data)
  File "/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/idf_monitor.py", line 387, in handle_serial_input
    if self._output_enabled and (self._force_line_print or self._line_matcher.match(line)):
  File "/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/idf_monitor.py", line 255, in match
    m = self._re.search(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [/home/justin/dev/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1
'make monitor' FAILED!
```

yeyeto2788, 2018-10-08:
Hello @superquest,

I would suggest that after installing the firmware onto the board try running the example scripts with [Adafruit's ampy tool](https://github.com/adafruit/ampy)

To flash the firmware first erase flash memory by typing `esptool.py --chip esp32 -p /dev/tty.SLAB_USBtoUART erase_flash` where the `/dev/tty.SLAB_USBtoUART` should be your serial port, after successfully erasing flash try to flash the firmware with this [info](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares).

Take into account that the `flash.sh` is using the `/dev/ttyUSB0` port to write the firmware into the board, so you might need to do something like this `../flash.sh -p <your_comm_port> -b <baud_rate>`

NOTE: remember to update the `esptool.py` to latest version.

Hope it helps.

superquest, 2018-10-08:
Thanks @yeyeto2788.

What do you mean by "your serial port".

I interpreted it to be `/dev/ttyUSB0` because that shows up when I plug the m5stack in.

But when I run:
```
esptool.py --chip esp32 -p /dev/ttyUSB0 erase_flash
cd firmware/esp32
../flash -p /dev/ttyUSB0
ampy -p /dev/ttyUSB0 ls
```
it just hangs ...

yeyeto2788, 2018-10-08:
If you look at the output for `dmesg | grep tty` you should see the output with some serial ports, one of those should be the one of the board. Depending on the board you might need also to install drivers for the USB to serial converter. In some cases, it is used the CH340 or the CP2102 or another USB to serial converted chip.

What kind of board are you using?

superquest, 2018-10-08:
Update: after hitting the reset button on my [**m5stack**](https://www.amazon.com/MakerFocus-Development-Bluetooth-Expandable-Compatible/dp/B075FRWH4S) I now get this output

```
$ ampy -p /dev/ttyUSB0 ls
flash
```

But picocom still hangs. I've gotten picocom to work before with an esp8266 board so I know how that should work.

@yeyeto2788: yes it is `/dev/ttyUSB0` then. That's the only candidate that shows up after executing the `dmesg | grep tty`.

So -- how might I find the needed drivers to talk to [**m5stack**](https://www.amazon.com/MakerFocus-Development-Bluetooth-Expandable-Compatible/dp/B075FRWH4S) from my Arch Linux machine?

Thanks again for your kind help :)

yeyeto2788, 2018-10-08:
From the internet I have found this [link](http://www.m5stack.com/assets/docs/) it seems to have the CP2104 USB to serial chip.

The weird thing is that if `ampy` got connected to the board showing the `flash` directory, so it is not something related to the serial port neither the drivers.

It seems that you are trying to connect to the board through `picocom` at baud rate different than the one set on firmware which is 115200.

Try with `picocom -b 115200 /dev/ttyUSB0`

superquest, 2018-10-08:
> Try with picocom -b 115200 /dev/ttyUSB0

That worked! Hallelujah!!!

The logging here could be improved:

```
$ ../flash.sh
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 84:0d:8e:21:10:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 21232 bytes to 12495...
Wrote 21232 bytes (12495 compressed) at 0x00001000 in 0.3 seconds (effective 572.5 kbit/s)...
Hash of data verified.
Compressed 144 bytes to 69...
Wrote 144 bytes (69 compressed) at 0x0000f000 in 0.0 seconds (effective 309.8 kbit/s)...
Hash of data verified.
Compressed 1539680 bytes to 930617...
Wrote 1539680 bytes (930617 compressed) at 0x00010000 in 22.2 seconds (effective 555.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 126...
Wrote 3072 bytes (126 compressed) at 0x00008000 in 0.0 seconds (effective 5722.2 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.
```

That output is why I was using the wrong baud rate.

@ @yeyeto2788 Now, how can I run [the Paint example](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/fede6f610c9952c7242fc309c2bb4efc9afb0919/MicroPython_BUILD/components/micropython/esp32/modules_examples/tft/paint.py) or another example like it? Just trying to get something to show up on my screen ...

yeyeto2788, 2018-10-08:
@superquest you should be able to do it, before doing so remember that the m5stack does not have touch controller.

I would suggest to try the [tftdemo.py](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/fede6f610c9952c7242fc309c2bb4efc9afb0919/MicroPython_BUILD/components/micropython/esp32/modules_examples/tft/tftdemo.py) as is since it have the `tft` setup for the m5stack.

There are also another scripts that you can use like these [scripts](https://github.com/yeyeto2788/MicroPythonScripts).

As far as I know there is no way to disable those logs since those are boot logs (not an expert :disappointed: )

Regards.

superquest, 2018-10-08:
But what do I do with those files? How do I put them on the board / run them?

justinmoon, 2018-10-09:
OK I figured out a good way. I basically forked https://github.com/tuupola/micropython-m5stack for my actual application.

closed 2018-10-09 by justinmoon