loboris/MicroPython_ESP32_psRAM_LoBo #228: support SH1106 OLED displays

jasonblewis, 2018-10-08:
Is it possible to support SH1106 OLED displays? The TTGO T-Eight board has one built into it.

If I try and use it by using the built in SD1306 driver, I get a few lines at the top of the screen and the rest of the screen looks like static.

It seems very similar to this issue, which might hold some clues as to how it could be supported:

https://github.com/rm-hull/luma.oled/issues/1

[image]

Also see https://github.com/adamyoung600/WRX_HUD/tree/master/Hardware/SH1106

```
import machine, time
from ssd1306 import SSD1306_I2C

WIDTH = const(132)
HEIGHT = const (64)
sda_pin = 21
scl_pin = 22

i2c = machine.I2C(scl=scl_pin, sda=sda_pin, speed=400000)

ssd = SSD1306_I2C(WIDTH, HEIGHT, i2c)

import freesans20

from writer import Writer
wri2 = Writer(ssd, freesans20, verbose=True)

Writer.set_clip(True, True)

for y in range(0,HEIGHT-1):
    for x in range(0,WIDTH-1):
        Writer.set_textpos(x, y)
        wri2.printstring('{} {} MicroPython\nby LoBo\n10/2017'.format(x,y))

        ssd.show()
        time.sleep(1)
    

```

carterw, 2018-10-08:
Do you have a pinout diagram for the TTGO board? I was experimenting with one a few months ago (different display dimensions but most likely very similar otherwise) and the I2C pins connected to the display were not 21 and 22. SCL was 15 and SDA was 4. 

See if the device shows up when you do i2c.scan().

i2c = machine.I2C(scl=machine.Pin(scl_pin_id), sda=machine.Pin(sda_pin_id), freq = freq)
devices = i2c.scan()
print("i2c scan", devices)

jasonblewis, 2018-10-08:
@carterw thanks, the board has a silkscreen on it saying SDA=21 and SCL=22, which is what I tested with, and got the result in the photo above. So I'm pretty confident they are the correct pins, its just not being driven correctly

carterw, 2018-10-08:
Well that's good, and I assume you did the scan and the device shows up in the results? There may be a reset pin attached to the display that has to be toggled, check for that. Also look in the driver for places where a read of a register happens and put a print statement there to see if you are actually getting data.

jasonblewis, 2018-10-09:
@loboris thank you! that works!

mikeskadavid99, 2021-08-22:
Dobrý den, mohu pomoci deska TTGA ESP8266 SH1106 (https://www.laskarduino.cz/lilygo-ttgo-to-esp8266-sh110-1-3--oled-meteostanice--wifi-modul/ ). stahuj ESP8266 weatherstation (DS1306) = pocasi .  upraveni SH1106. zapis a neukazuje obraz. neco chyba posim pomoc
[esp8266-weather-station.zip](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/7027790/esp8266-weather-station.zip)(original) prosim prelozit. Dik DM email:mikeskadavid@tiscali.cz