loboris/MicroPython_ESP32_psRAM_LoBo #230: ESP32 OdroidGo: Error initializing WiFi (4353)

jedie, 2018-10-09:
```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
W (11939) wifi: wifi osi_nvs_open fail ret=4353

E (11940) wifi: wifi_init_in_caller_task 707 ret=4353
[0;31mE (11940) [modnetwork]: Error initializing WiFi (4353)[0m
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 4353
```

capedra, 2018-10-10:
> ```
> W (11939) wifi: wifi osi_nvs_open fail ret=4353
>
> E (11940) wifi: wifi_init_in_caller_task 707 ret=4353
> �[0;31mE (11940) [modnetwork]: Error initializing WiFi (4353)�[0m
> Traceback (most recent [...]

https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/209

jedie, 2018-10-10:
@capedra Thanks!

Yes, it worked, after i have run `esptool.py erase_flash` and reflash the firmware ;)

Maybe it's because there was mainline microPython flashed before and I probably didn't do an erase.

loboris, 2018-10-11:
ESP32 WiFi driver saves the configuration in NVS and a conflicting configuration may exist after esp-idf version change.
It is not releted only to MicroPython.

jedie, 2018-10-11:
Thanks for the info. What's about to add this information into wiki ?