loboris/MicroPython_ESP32_psRAM_LoBo #232: esp32_psram_all_bt missing on wiki/firmwares

jedie, 2018-10-10:
There exists: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/firmware/esp32_psram_all_bt

But this is missing on wiki page: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares

and it's missing in README: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/firmware/README.md

I guess "bt" stands for "Bluetooth" isn't it?
Only `esp32_psram_all_bt` has `CONFIG_BT_ENABLED=y` in sdkconfig...

loboris, 2018-10-11:
Thank you, I'll update the Wiki.

jedie, 2018-10-11:
Btw, what's about to open the wiki edits for everybody?

loboris, 2018-10-12:
I'll think about it.
In the meantime, you can always suggest some changes or write a full article and post it here, on the forum, as a PR or as a private message...