loboris/MicroPython_ESP32_psRAM_LoBo #233: Camera support.

beyonlo, 2018-10-11:
Hello.

Are there plans to support Camera using I2S on MicroPython?

There are some ESP32 module that supports OV2640 and OV7670 cameras, Built-in Flash lamp.

https://www.seeedstudio.com/ESP32-CAM-Development-Board(with-camera)-p-3153.html
https://www.banggood.com/M5Stack-Official-ESP32-Camera-Module-Development-Board-OV2640-Camera-Type-C-Grove-Port-p-1333598.html?cur_warehouse=CN

Thank you.

loboris, 2018-10-12:
The support is planned and should be ready in a couple of months.

beyonlo, 2018-10-15:
Hi Loboris.

That is a great news. =D
Will you use I2S for that?

Thank you.

realfogman, 2019-11-03:
Is there any news regarding camera support?
It seems, that commits have stopped a while ago?

chmondkind, 2019-11-03:
Sorry mate - this fork, the one that is „the“ fork for Espressif‘s MCU‘s is dead.
I don’t give Boris any fault for this, obviously one has an income during such extensive work and uses the modules professionally or for an uni degree.
Official uPython is integrating some parts of this fork, check their forum and that on LoBo‘s website.

realfogman, 2019-11-04:
Hello chmondkind,

the link is something I was reading before and my thought was, that "The camera module will be commited in a few days.".
I do fully understand, that such a project takes a huge amount of resources and I can only ask to commit the camera code in its current state. (Although it may not work correctly.)
This might be helpful for others to support the project in regard of fixing bugs etc.

chmondkind, 2019-11-04:
Hi realfrogman - If it isn’t there are viable alternatives working on K210 boards.

I‘ve ordered one to test its automated visual inspection capabilities. It has been done on this platform, with python.