loboris/MicroPython_ESP32_psRAM_LoBo #236: How to use SDCard in SPI mode & spi_master at the same time on the M5Stack?

Abol-Wang, 2018-10-23:


loboris, 2018-10-25:
It there any specific issue? It should work.

Abol-Wang, 2018-10-25:
@loboris I use esp-idf with M5STACK, and I wonder how you can implement LCD and SD in mpy to work together？
can you separate your solution from mpy, and synchronize the esp32 TFT driver of esp-idf by the TFT driver lasted code in mpy.

Abol-Wang, 2018-10-25:
I don't think this old TFT driver library works very well. Can you update it to the latest version with color565？
https://github.com/loboris/ESP32_TFT_library

loboris, 2018-10-25:
SDCard in SPI mode and LCD on M5Stack works good in MicroPython.
I had to modify some esp-idf drivers (spi-master & sdspi_host) to make it work.

I'll try to update https://github.com/loboris/ESP32_TFT_library if I'll find some time.

Abol-Wang, 2018-10-25:
@loboris Thank you very much，wait for your good news quietly！