loboris/MicroPython_ESP32_psRAM_LoBo #243: How to include main.py and boot.py

devmare, 2018-11-30:
I'm not able to include **main.py** and **boot.py** file in the firmware image. I tried several locations and also created scripts folders but no success. I'm only able to include .py files in the modules folder.

According to the code following folders are used:
FROZEN_DIR = $(COMPONENT_PATH)/esp32/scripts
FROZEN_MPY_DIR = $(COMPONENT_PATH)/esp32/modules

Therefore I also tried following path:
**MicroPython_ESP32_psRAM_LoBo\MicroPython_BUILD\components\micropython\esp32\scripts\main.py**

Is there something else I have to consider?

carterw, 2018-12-01:
Here's what I do;

```
export LOBO="${HOME}/esp32/LoBo
export MODULES="${LOBO}/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modules"
export FSIMAGE="${LOBO}/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/internalfs_image/image"

MFILES="your comma-separated list of files or directories to embed into the firmware"
FSFILES="your comma-separated list of files or directories to show up in the filesystem"

cp -r ${MFILES} ${MODULES}
cp -r ${FSFILES} ${FSIMAGE}
```
Then you have to build/flash the firmware image and then separately do the same for the filesystem image.

loboris, 2018-12-01:
You can simply replace **main.py** and/or **boot.py** in:
[MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/internalfs_image/image/](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/master/MicroPython_BUILD/components/internalfs_image/image)
and then [prepare and flash the filesystem](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/filesystems#using-prepared-file-system-images).

devmare, 2018-12-01:
I already tried this. I also removed the example stuff but this didn't change anything.
I only added a print command to main.py and boot.py but no output appears.
Here is what I get now (output is shortened to important parts):

`./BUILD.sh makefs`
Creating SPIFFS image...
Making spiffs image; Flash address: 0x2b0000, Size: 1344 KB ...
/boot.py
/main.py
/spiffs.info
/mytest.py
To flash to ESP32 execute:
python ...
or execute ./BUILD.sh flashfs
OK

`./BUILD.sh flashfs`
Flashing SPIFFS image to ESP32...
Making spiffs image; Flash address: 0x2b0000, Size: 1344 KB ...
/boot.py
/main.py
/spiffs.info
/mytest.py
Flashing the image ...
Chip is ESP32D0WDQ6 (revision 1)
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1376256 bytes to 3330...
Wrote 1376256 bytes (3330 compressed) at 0x002b0000 in 0.0 seconds (effective 282344.9 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
OK.

`./BUILD.sh monitor` and pressed reset button
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4820
load:0x40078000,len:8652
load:0x40080000,len:5748
0x40080000: _WindowOverflow4 at .../esplobodir/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x400802dc
0x400802dc: _NMIExceptionVector at ??:?

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 983040; Flash address: 0x310000]
Filesystem size: 896256 B
           Used: 512 B
           Free: 895744 B
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.

`import os; os.listdir()`
['boot.py']

loboris, 2018-12-01:
I've tried it and it works.
It looks like something is wrong with your partition table. The file system was flashed at 0x002b0000 and the FS was mounted at 0x310000.

Please try to execute the following (you may use `./BUILD.sh -j8 all` for faster building):
```
./BUILD.sh clean
./BUILD.sh all
./BUILD.sh flash
./BUILD.sh flashfs

./BUILD.sh monitor
```
To prevent reset after flash commands set in mennuconfig:
`→ Serial flasher config → After flashing (Stay in bootloader)`

loboris, 2018-12-01:
Never edit partition table (`partitions_mpy.csv`) manualy.

If you don't want the partition sizes to be determined automatically, you may use [BUILD.sh](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/build#options) options.
Usualy, it is enough to use **-a** option to set the application partition size.
File system size option **-fs** is mostly used to **limit** the file system size (if you don't want it to fill all the remaining Flash).

The **final** partitions table is generated when **`./BUILD.sh flash`** command is executed, when the final firmware size is known. All partition sizes are checked and warning is issued if the firmware does not fit in Flash size.

After building the firmware you may execute **flash** command with **dummy port** selected, correct partitions table will be generated, but no actual flasshing will be performed.
```./BUILD.sh -a 1500 -p dummy flash```
After the command is executed you may check the `partitions_mpy.csv` to check what will be flashed.

devmare, 2018-12-01:
Thanks so much for you support, everything works now. There was also a problem with building mkspiffs in build_func.sh `make -C components/mkspiffs > /dev/null 2>&1`

Building mkspiffs
FAILED
'make makefs' FAILED!

This was because g++ was missing but there was no detailed error message. It would be better to remove `2>&1`. Also g++ package requirement should be added for Ubuntu.