loboris/MicroPython_ESP32_psRAM_LoBo #244: [Wiki] - rtc.read() and rtc.read_string() have incorrect return type

chrisb2, 2018-12-01:
In the Wiki for the machine.RTC class.

[rtc.read()](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/rtc#rtcreadpos) should say:
> Read integer (32-bit) from the position pos in RTC memory.
> Return an integer if one has been written to the position, or _None_ otherwise.

[rtc.read_string()](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/rtc#rtcread_string) should say:
> Read the string from RTC memory.
> Return a string if one has been written to the position, or _None_ otherwise.

loboris, 2018-12-01:
Thank you, the Wiki is [updated](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/rtc#rtcreadpos).
The acctual behaviour is slightly different than your observation.

chrisb2, 2018-12-01:
Thank you, thats really clear.

closed 2018-12-01 by chrisb2