loboris/MicroPython_ESP32_psRAM_LoBo #251: SD card can't be mounted anymore

fivdi, 2018-12-15:
On an ESP-WROVER-KIT V3 it's possible to access an SD card with MicroPython ESP32_LoBo_v3.2.20. This can be seen here:
```
MicroPython ESP32_LoBo_v3.2.20 - 2018-06-28 on ESP32 board with ESP32
Type "help()" for more information.
>>> import uos
>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (1bit)
 Name: 00000
 Type: SDHC/SDXC
Speed: high speed (50 MHz)
 Size: 15423 MB
  CSD: ver=1, sector_size=512, capacity=31586304 read_bl_len=9
  SCR: sd_spec=2, bus_width=5

>>> uos.listdir("/sd")
['FOO.TXT', 'hello']
>>> 
```
If MicroPython ESP32_LoBo_v3.2.24 is used on the same ESP-WROVER-KIT the SD card can no longer be accessed and the CPU is halted:
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32&psRAM board with ESP32
Type "help()" for more information.
>>> import uos
>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (1bit)
     Name: 00000
     Type: SDHC/SDXC
    Speed: high speed (50 MHz)
     Size: 15423 MB
      CSD: ver=1, sector_size=512, capacity=31586304 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009e9a0  PS      : 0x00060f30  A0      : 0x80103a54  A1      : 0x3ffdbb30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0xfffffffd  A5      : 0x00000000  
A6      : 0x3f415b44  A7      : 0x3f415b44  A8      : 0x00000000  A9      : 0x00000073  
A10     : 0x3f4178f0  A11     : 0x00000001  A12     : 0x3f813301  A13     : 0x3ffdbb44  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4009e911  LEND    : 0x4009e921  LCOUNT  : 0xffffffff  

Backtrace: 0x4009e9a0:0x3ffdbb30 0x40103a51:0x3ffdbb50 0x40103ade:0x3ffdbb70 0x400f585a:0x3ffdbbb0 0x400f1231:0x3ffdbbe0 0x40104d7c:0x3ffdbc00 0x4010585e:0x3ffdbc40 0x400f5866:0x3ffdbc60 0x400f1231:0x3ffdbc90 0x400f1299:0x3ffdbcb0 0x400fe566:0x3ffdbcd0 0x400f58f0:0x3ffdbd70 0x400f1231:0x3ffdbdd0 0x400f125e:0x3ffdbdf0 0x400e2d24:0x3ffdbe10 0x400e2fa0:0x3ffdbec0 0x400d5d9c:0x3ffdbf00

CPU halted.
```

loboris, 2018-12-17:
I can't reproduce this.
Running on an ESP-WROVER-KIT V3:
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>> import uos
>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (1bit)
     Name: SL08G
     Type: SDHC/SDXC
    Speed: high speed (50 MHz)
     Size: 7580 MB
      CSD: ver=1, sector_size=512, capacity=15523840 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

>>> 
>>> uos.listdir('/sd')
['test1.avi', 'foo.txt', 'Picture', 'Video', 'Audio', 'VideoCall', 'Money.wav', 'Wall1.raw']
>>> 
```

Please, run with `./BUILD.sh monitor` to get the the full crash trace log.

BTW, on ESP-WROVER-KIT V3 you can use SD Card in 4-line SD mode.

fivdi, 2018-12-17:
It looks like we tested with different variants of the ESP32_LoBo_v3.2.24 firmware. The first line of output from your test output is:
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
```
The first line of output from my test output is:
```
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32&psRAM board with ESP32
```
I tested with the [firmware](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares) called "esp32_psram_all_bt" to have psRAM available. I think you tested with the firmware called "esp32_all". "esp32_all" also works for me but then psRAM isn't available.

Should the SD Card interface work in mode SDMODE_1LINE and SDMODE_4LINE wmen psRAM is enabled?

fivdi, 2018-12-18:
Here's the output of `./BUILD.sh monitor`:

```
$ ./BUILD.sh monitor

---------------------
MicroPython for ESP32
---------------------

============================
Executing esp-idf monitor...
============================
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5868
load:0x40078000,len:7756
ho 0 tail 12 room 4
load:0x40080000,len:6016
0x40080000: _WindowOverflow4 at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080324
0x40080324: _KernelExceptionVector at ??:?

I (30) boot: ESP-IDF v3.1-rc1-53509c7b 2nd stage bootloader
I (31) boot: compile time 13:17:40
I (50) boot: Enabling RNG early entropy source...
I (51) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (53) boot: SPI Flash Size : 4MB
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 MicroPython      factory app      00 00 00010000 00240000
I (90) boot:  3 internalfs       Unknown data     01 82 00250000 00100000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x70d28 (462120) map
I (272) esp_image: segment 1: paddr=0x00080d50 vaddr=0x3ffc0000 size=0x05808 ( 22536) load
I (281) esp_image: segment 2: paddr=0x00086560 vaddr=0x3ffc5808 size=0x00000 (     0) load
I (281) esp_image: segment 3: paddr=0x00086568 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

I (288) esp_image: segment 4: paddr=0x00086970 vaddr=0x40080400 size=0x096a0 ( 38560) load
I (313) esp_image: segment 5: paddr=0x00090018 vaddr=0x400d0018 size=0x19eb4c (1698636) map
0x400d0018: _flash_cache_start at ??:?

I (907) esp_image: segment 6: paddr=0x0022eb6c vaddr=0x40089aa0 size=0x16034 ( 90164) load
0x40089aa0: ppTxPkt at ??:?

I (944) esp_image: segment 7: paddr=0x00244ba8 vaddr=0x400c0000 size=0x00664 (  1636) load
I (944) esp_image: segment 8: paddr=0x00245214 vaddr=0x50000000 size=0x0098c (  2444) load
I (971) boot: Loaded app from partition at offset 0x10000
I (971) boot: Disabling RNG early entropy source...
I (972) spiram: SPI RAM mode: flash 40m sram 40m
I (976) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (983) cpu_start: Pro cpu up.
I (987) cpu_start: Starting app cpu, entry point is 0x400816cc
0x400816cc: esp_intr_disable at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/intr_alloc.c:768

I (0) cpu_start: App cpu up.
I (1848) spiram: SPI SRAM memory test OK
I (1849) heap_init: Initializing. RAM available for dynamic allocation:
[... 70 lines cut ...]

0x4010585e: uart_consume at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modymodem.c:110

0x400f5866: instance_getiter at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:872

0x400f1231: fun_bc_call at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:274

0x400f1299: fun_bc_call at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:335

0x400fe566: mp_vfs_getcwd at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/../extmod/vfs.c:307

0x400f58f0: super_attr at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:1147

0x400f1231: fun_bc_call at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:274

0x400f125e: fun_bc_call at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:283

0x400e2d24: ftp_isenabled at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c:1393

0x400e2fa0: _init at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ow/owb_rmt.c:380

0x400d5d9c: machine_pin_obj_init at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/machine_pin.c:473

CPU halted.
```

loboris, 2018-12-18:
@fivdi
There is no difference in SD Card operations with or without psRAM.

Your log shows that the crash was not caused by sdcard related code.

fivdi, 2018-12-18:
>  There is no difference in SD Card operations with or without psRAM.

Thank you for confirming this.

> Your log shows that the crash was not caused by sdcard related code.

The firmware on the board was downloaded from this repository and flashed onto the board. The firmware wasn't modified in any way.

Can you by any chance reproduce the error with the `esp32_psram_all_bt` variant of `MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06`? If you can't reproduce the error with `esp32_psram_all_bt` do you have a suggestion as to how I can proceed?

loboris, 2018-12-18:
Ok, I'll test it later today.
Have you tested with the [basic psRAM firmware](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/raw/master/MicroPython_BUILD/firmware/MicroPython_LoBo_esp32_all.zip) (the one without BT support) ?

fivdi, 2018-12-18:
> Ok, I'll test it later today.

Excellent, thank you very much.

> Have you tested with the [basic psRAM firmware](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/raw/master/MicroPython_BUILD/firmware/MicroPython_LoBo_esp32_all.zip) (the one without BT support) ?

The file linked to here is called `MicroPython_LoBo_esp32_all.zip` and to the best of my knowledge it doesn't support psRAM. If I understand the firmware naming conventions correctly the firmware without BT support that does support psRAM is [here](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/firmware/MicroPython_LoBo_esp32_psram_all.zip) and is called `MicroPython_LoBo_esp32_psram_all.zip`.

Just to be sure I tested with the firmware in both `MicroPython_LoBo_esp32_all.zip` and `MicroPython_LoBo_esp32_psram_all.zip`. The firmware in `MicroPython_LoBo_esp32_all.zip` can successfully access the SD card. The firmware in `MicroPython_LoBo_esp32_psram_all.zip` can't successfully mount the SD card.

If the firmware which doesn't support psRAM in file `MicroPython_LoBo_esp32_all.zip` is used the SD card can be successfully accessed. Here is the output of `./BUILD.sh monitor`:
```
$ ./BUILD.sh monitor

---------------------
MicroPython for ESP32
---------------------

============================
Executing esp-idf monitor...
============================
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5976
load:0x40078000,len:9040
ho 0 tail 12 room 4
load:0x40080000,len:6120
0x40080000: _WindowOverflow4 at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080334
0x40080334: _KernelExceptionVector at ??:?

I (30) boot: ESP-IDF v3.1-rc1-53509c7b 2nd stage bootloader
I (31) boot: compile time 13:16:14
I (48) boot: Enabling RNG early entropy source...
I (48) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (51) boot: SPI Flash Size : 4MB
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 MicroPython      factory app      00 00 00010000 001f0000
I (88) boot:  3 internalfs       Unknown data     01 82 00200000 00100000
I (95) boot: End of partition table
I (100) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x735b8 (472504) map
I (274) esp_image: segment 1: paddr=0x000835e0 vaddr=0x3ffb0000 size=0x05040 ( 20544) load
I (282) esp_image: segment 2: paddr=0x00088628 vaddr=0x3ffb5040 size=0x00000 (     0) load
I (283) esp_image: segment 3: paddr=0x00088630 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

I (290) esp_image: segment 4: paddr=0x00088a38 vaddr=0x40080400 size=0x075d8 ( 30168) load
I (311) esp_image: segment 5: paddr=0x00090018 vaddr=0x400d0018 size=0x154b7c (1395580) map
0x400d0018: _flash_cache_start at ??:?

I (800) esp_image: segment 6: paddr=0x001e4b9c vaddr=0x400879d8 size=0x0d92c ( 55596) load
0x400879d8: cnx_node_search at ??:?

I (823) esp_image: segment 7: paddr=0x001f24d0 vaddr=0x400c0000 size=0x00648 (  1608) load
I (824) esp_image: segment 8: paddr=0x001f2b20 vaddr=0x50000000 size=0x0098c (  2444) load
I (845) boot: Loaded app from partition at offset 0x10000
I (845) boot: Disabling RNG early entropy source...
[... 200 lines cut ...]

0x4010bbca: sprintf at ??:?

0x400fa83f: str_startswith_word at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/repl.c:37 (discriminator 2)

0x400f5acd: mp_obj_instance_load_attr at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:606

0x400f5b3d: mp_obj_instance_call at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:840

0x40103e9a: machine_uart_init_helper at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/machine_uart.c:588

0x400fa8da: mp_repl_continue_with_input at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/repl.c:62

0x400f5acd: mp_obj_instance_load_attr at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:606

0x400f5afa: mp_obj_instance_load_attr at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objtype.c:646

0x400e56e3: mp_print_mp_int at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mpprint.c:220

0x400e5981: mp_print_float at ??:?

0x400d7fb0: _esp_exceptions at /home/brian/src/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:91

CPU halted.
```

closed 2019-03-30 by fivdi