loboris/MicroPython_ESP32_psRAM_LoBo #257: Sleep support (5 sleep mode).

beyonlo, 2019-01-02:
Hi, Is that 5 sleep mode supported by the Lobo MicroPython?

https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/

Thank you.

carterw, 2019-01-02:
From what I can tell only one sleep mode seems to have been implemented, and I think it is the 'deep sleep'. Power draw is in the vicinity of 10µA.

https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/rtc

L4grange, 2019-05-30:
Does anyone know if the esp module is implemented in the LoBo firmware?
The MicroPython docs here http://docs.micropython.org/en/latest/library/esp.html state that the Modem Sleep sleep mode is supported, but I cannot find the esp module in the LoBo firmware.

carterw, 2019-05-31:
Doesn't appear to have been implemented. Possibly will never be, the last commit was Sept. 2018.