loboris/MicroPython_ESP32_psRAM_LoBo #260: Error on enable support for SPI TFT displays

GustavoRSSilva, 2019-01-06:
I am trying to enable support for SPI TFT displays.

Everything seems to work properly expect when I enable this option.

The error is:

In file included from /Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:33:0:
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:48:50: error: 'MP_QSTR_display' undeclared here (not in a function)
	     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_display) },
	                                                  ^
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/obj.h:93:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
	 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
	                                                        ^
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:48:38: note: in expansion of macro 'MP_ROM_QSTR'
	     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_display) },
	                                      ^
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:48:5: warning: missing initializer for field 'value' of 'mp_rom_map_elem_t {aka const struct _mp_rom_map_elem_t}' [-Wmissing-field-initializers]
	     { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_display) },
	     ^
	In file included from /Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:33:0:
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/obj.h:341:18: note: 'value' declared here
	     mp_rom_obj_t value;
	                  ^
	In file included from /Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:33:0:
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/moddisplay.c:51:23: error: 'MP_QSTR_TFT' undeclared here (not in a function)
	     { MP_OBJ_NEW_QSTR(MP_QSTR_TFT), MP_ROM_PTR(&display_tft_type) },
	                       ^
	/Users/me/workstation/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/obj.h:93:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
	 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))

milinatw, 2020-11-28:
https://loboris.eu/forum/showthread.php?tid=111 shows following, and it works for me.

After changing the configuration via menuconfig you should (almost always) clean the build first:

./BUILD clean

or to clean and build:
./BUILD clean all

or for faster build:
./BUILD -j4 clean all

patbeppo, 2021-01-21:
I have the exact same problem. I tried ./BUILD clean: no success.
milinatw: could you please indicate what options you chose in menuconfig. Might there be a dependency?

patbeppo, 2021-01-22:
After deselecting all modules, build clean (without any -j) and selecting it again it worked.