loboris/MicroPython_ESP32_psRAM_LoBo #263: QUESTION: documentation add a new c-module

ahaeberle, 2019-01-13:
I tried to add a c-module to my project but the documentation https://micropython-dev-docs.readthedocs.io/en/latest/adding-module.html does not fit.

I am still hanging on the very first step, adding a module (like esp32/mymodule.c)
If I add it to the components.mk in `MicroPython_BUILD/components/` the build script will tell me `*** No rule to make target`

There is still no code in mymodule.c !

Sincerly Felix

MikeTeachman, 2019-01-17:
Here is a working example on how to create a C module on the LoBo port.  If you make all the changes in the commit you should get a working "argex" module that has examples on how to expose your C code as micropython function and class methods.  From there, extend the example with your code.  good luck !
https://github.com/MikeTeachman/MicroPython_ESP32_psRAM_LoBo_I2S/commit/c92e00f7590b84eaf5f9818b1458c478d0b2cadf

vrubel, 2019-01-17:
https://github.com/vrubel/MicroPython_ESP32_psRAM_LoBo/commit/57227919ffa93f426fb21212a5d949ea92f2d9f5

ahaeberle, 2019-01-26:
Hey thanks a lot that helped. Will study the diff and try.
(Find a nice feature of visual studio code regarding the russian comments - feels like future if the computer can translate this in an application so quickly...)
Felix

lagmoellertim, 2020-03-26:
Hey, I tried to include the ulab library (written for micropython) in this fork. 
I always get this error:
`error: conflicting type qualifiers for 'ulab_user_cmodule'
 mp_obj_module_t ulab_user_cmodule`

As well as `previous declaration of 'ulab_user_cmodule' was here
 extern const struct _mp_obj_module_t ulab_user_cmodule;`

I followed the exact directions from your commit, but it still didn't help.
Does anyone know what I'm doing wrong?

mdaeron, 2020-07-26:
> Hey, I tried to include the ulab library (written for micropython) in this fork.
> I always get this error:
> `error: conflicting type qualifiers for 'ulab_user_cmodule' mp_obj_module_t [...]

Hi, were you eventually able to compile ulab?