loboris/MicroPython_ESP32_psRAM_LoBo #266: LittlevGL and μGFX support for uPython on ESP32.

beyonlo, 2019-01-21:
Hi all!

1. I would like to discuss with you about that great GUI libs for ESP32, what you think about?
2. Is a good idea to support it on uPython on ESP32?
3. Is need a hard work to support it? Please, correct me if I'm wrong, but if that lib is pure C, to do a interface to uPython is not so hard, right? How that works?

Thank you.

amirgon, 2019-01-21:
@beyonlo  See this: https://github.com/littlevgl/lvgl/issues/557

ril3y, 2019-07-30:
Any change we can add lvgl to this port of python?

amirgon, 2019-07-30:
> Any change we can add lvgl to this port of python?

@ril3y lvgl already supports Micropython.  

You can either use [lv_micropython](https://github.com/littlevgl/lv_micropython) directly, or intergate lvgl to some other Micropython fork (such as this one) with a small effort.