loboris/MicroPython_ESP32_psRAM_LoBo #268: mqtt - best way to subscribe to several topics?

dubaleeiro, 2019-02-02:
Hello @loboris ,

first of all thanks a lot for the great repository ! This is actually more like a question than a issue.

My current approach to subscribe to the following mqtt topics is just using the `mqtt.subscribe` method one after the other, with a 3 sec delay between then.
'/plant/temp'
'/plant/humi'
'/plant/status'

I have noticed that if I symply subscribe to the topic '/plant/', the websockets monitor at cloud.mqtt.com website display the messages, but this is not the case with the callback `data_cb`, which only displays if the topic is exactly '/plant/'. Is there any way to change that?

Thanks !
Eduardo

infogeek, 2019-02-21:
Hi @dubaleeiro,
I think you should be able to do this using 
'plant/#'
I have tested it on the unix port of micropython with umqtt.simple.

dubaleeiro, 2019-02-24:
Hello @infogeek ,
Thanks for the great tip. It worked!
However, when the subscription using 'plant/#' is confirmed, I get the retained message of only 1 out of 3 topics I have previously published and retained into the mqtt server. Would you have any hints on that as well?
Thanks!