loboris/MicroPython_ESP32_psRAM_LoBo #277: import sx127x doesn't work

johncblacker, 2019-02-19:
If I create a /flash/lib/lora directory and put sx127x.py in that directory, the 
import sx127x
doesn't work.  I thought I can import modules from anywhere in the file system as indicated by doing a
help('modules') - the last line says: "Plus any modules on the filesystem"
Am I misunderstanding how things work?

ucod, 2019-02-23:
you're not pointing to the containing folder, but rather top level.
try
>from flash.lib.lora import sx127x 
you can import whatever as long as it's on the same level or with path

johncblacker, 2019-03-07:
Thanks for the help.  I'm encountering another problem.  I have two ttgo boards, one is running your esp32 firmware and the other is running firmware I built from the original MP repository.  I'm trying to get the LightLora programs to work (github LightLora) and the program running on the board with your repository, seems to be "stuck" as it prints out it's first receive message and then just sits there.  The app running on the ttgo with the repository that I built from the MP repository seems to be working fine.  So, I flashed my firmware onto the ttgo that had your repo, just to see if there might be some kind of board issue and now the apps on both boards are happily sending and receiving.  So, I can only conclude there's something going on with your distro or the firmware I used.  I flashed the firmware from:
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/firmware/MicroPython_LoBo_esp32_psram.zip I did get a message indicating the no SPIRAM was detected, but otherwise everything else seems to be OK.  Any ideas?  What can I do to diagnose the problem?

johncblacker, 2019-03-07:
Just for grins, I downloaded the firmware version without PSRAM, flashed the board again and ran my LightLora test again and got same results.