loboris/MicroPython_ESP32_psRAM_LoBo #282: Cannot build for OTA due to missing esp_http_client.h

gmos, 2019-03-01:
Hi Loboris,
Thanks for your splendid work. 

But I tried to compile for OTA on 4MB flash.  Straight clone on 1-3-19, nothing changed.
Fails with:

> ================================
> Building MicroPython firmware...
> ================================
> Building partitions from /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
> /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/component_wrapper.mk:213: warning: overriding recipe for target 'libmicropython.a'
> /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:132: warning: ignoring old recipe for target 'libmicropython.a'
> GENlast /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last
> /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/modota.c:49:29: fatal error: esp_http_client.h: No such file or directory
> compilation terminated.
> /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/mkrules.mk:75: recipe for target '/home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last' failed
> make[1]: *** [/home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last] Error 1
> make[1]: *** Deleting file '/home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last'
> /home/gijsmos/micropython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project.mk:468: recipe for target 'component-micropython-build' failed
> make: *** [component-micropython-build] Error 2
> 'make all' FAILED!

gmos, 2019-03-01:
Only happens if the "requests" module is deselected.

MuhammedImdaad, 2021-10-17:
were you able to solve this? I can import the client library if I am in the same directory with main. otherwise, it says cannot import?