loboris/MicroPython_ESP32_psRAM_LoBo #286: DNS Resolution for CURL and websockets

besi, 2019-03-11:
I am using the curl module and I ran into problems when it comes to DNS resolution

### Example 1 using curl

```python
import curl
curl.get('google.com')
(-5, '', '')
``` 

No data is loaded.

### Example 2: Using sockets:

I have a method called `http_get` which implements a http GET request in a minimal fashion.
Here again if I'm using an IP for the host it works fine. If I use a hostname it won't find the desired server:

````python
# This will work:
http_get("http://192.168.1.124/?play=1")

# The following won't work
http_get("http://myserver.local/?play=1")
````

The method:

```
def http_get(url):
    _, _, host, path = url.split('/', 3)
    addr = socket.getaddrinfo(host, 80)[0][-1]
    # The address can't be resolved

    s = socket.socket()
    s.settimeout(3)
    try:
        s.connect(addr)
        s.send(bytes('GET /%s HTTP/1.0\r\nHost: %s\r\n\r\n' % (path, host), 'utf8'))
        s.close()
    except:
        print("An exception occurred while loading the url %s" % url)
```

mrwhy-orig, 2019-03-20:
@besi did you build the fw or are you using a prebuild one?

besi, 2019-03-20:
It was built from the master branch