loboris/MicroPython_ESP32_psRAM_LoBo #289: Can't flash due to esp-idt submodule error

WayneKeenan, 2019-03-19:
The binary and filesystem built but I'm not able to flash using:

```
./BUILD.sh -v flash
```

Because of this error:
```
WARNING: Missing submodule components/expat/expat...
ERROR: esp-idf must be cloned from git to work.
'make flash' FAILED!
```

chmondkind, 2019-03-19:
This isn‘t an issue with this distro. It‘s an error that got thrown at you from esp-idf. as you could make it looks like there‘s something missing or the PATH env var isn‘t set/pointing where it should.
If you downloaded esp-idf, and in the right version (if there are constrictions), then try using esptool.
There are a load of tutorials how to do that out there

WayneKeenan, 2019-03-20:
I have a working 'esp-idf home' for another project and this distro ships an esp-idf zip and uses that under `Tools`.

I think the problem was I did a `make` without using the `BUILD.sh` which resulted in a confusing mix...

Resolved by a `make clean` and just using the `BUILD.sh` script.

closed 2019-03-20 by WayneKeenan