loboris/MicroPython_ESP32_psRAM_LoBo #291: upip error -29184

supersjimmie, 2019-03-27:
```
>>> upip.install('aenum')
Installing to: /flash/lib/
Error installing 'aenum': -29184, packages may be partially installed

```
I think it's not really directly related to this specific micropython, but I hope you can help me starting...  When a package-name is incorrect it says "Package not found" so I assume the name is correct, but how can I figure out why this strange -29184 error occurs?

curlyz, 2019-07-14:
it is the socket problem , possibly due to big socket.recv()

curlyz, 2019-07-14:
this happen due to ssl verification , in Python , we can disable this with adding verify = False inside requests method , not in micropython. :((

curlyz, 2019-07-16:
this is the server problem, sometimes they forgot to update their ssl certificate, please close this issue 
@supersjimmie

closed 2019-07-17 by supersjimmie